kizniche/Mycodo #1149: Pyrsistent fails to build during install leading to mycodo service not being able to start.

iinewmanii, 2022-02-02:
### Describe the problem/bug

During installation I receive an error that pyrsistent has failed to build. The installation will continue afterwards and will fail to connect with the daemon at the end. I have built pyrsistent outside of the install with no problems. I have searched around the github and tried various fixes but nothing helped so i decided to open an issue.

### Versions:

 - Mycodo Version: *8.12.9*
 - Raspberry Pi Version: *4*
 - Raspbian OS Version: *2022-01-28-raspios-bullseye-arm64*
 - Python Version: 3.9.2

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

It is a fresh install of raspian. All I did afterward was setup wlan and create an ssh file. I the did apt update/upgrade and then started trying to install Mycodo.

### Expected behavior

Expect to install without error.

Appreciate any help.

[setup.log](https://github.com/kizniche/Mycodo/files/7989428/setup.log)

kizniche, 2022-02-02:
Change the setuptools version in Mycodo/install/requirements.txt to setuptools==60.7.0 and rerun setup.sh and see if it fixes the issue.

iinewmanii, 2022-02-02:
Ok I tried that and I got the same result pyrsistent failed to build again. Could it be the version 0.18.1 of pyrsistent that is causing the problem?

kizniche, 2022-02-02:
I just pushed an update that pegs the setuptools version at 60.6.0, which is the previous version. You will need to pull the latest code from master with ```mv ~/Mycodo ~/Mycodo-bu && git clone https://github.com/kizniche/Mycodo && ~/Mycodo/install/setup.sh```

If that doesn't work, change this line:

https://github.com/kizniche/Mycodo/blob/db24ee8167bec9c70c15c4f97aa61c9e0ef49cc7/mycodo/scripts/upgrade_commands.sh#L22

And this line:

https://github.com/kizniche/Mycodo/blob/db24ee8167bec9c70c15c4f97aa61c9e0ef49cc7/install/requirements.txt#L30

to a previous version found at https://pypi.org/project/setuptools/#history and see if going back to a previous version allows it to install.

I just did a fresh install and am not having the same issue. Also, you neglected to mention the version of raspberry Pi OS, either Lite or Desktop.

iinewmanii, 2022-02-02:
I am using desktop version of 64-bit bullseye raspian.

iinewmanii, 2022-02-02:
I do not have that line in my upgrade_commands.sh file. Do I need to add that?

kizniche, 2022-02-02:
>I do not have that line in my upgrade_commands.sh file

Then that means you are not using the code on the master branch. The command I mentioned in https://github.com/kizniche/Mycodo/issues/1149#issuecomment-1028424386 instructs how to clone the Mycodo repo, which will obtain the code in the master branch.

iinewmanii, 2022-02-04:
I got the master branch code after you sent me this message. I tried different versions of setuptools for a while but kept getting the same results. I ended up erasing my sd card and installing the latest 32-bit version of raspberry pi os with desktop and then installing mycodo and didn't have any problems. So maybe it was something  with the 64-bit version that was causing an issue. I got it up and running now no problems so I am going to go ahead and start playing with it. Thank you for trying to help me out.

kizniche, 2022-02-05:
You're welcome. Glad you figured it out!

dylandn, 2022-02-09:
I'm hitting this bug as well, pyrsistent fails to build and mycodoflask.service fails to start. Also on a brand new 64bit install. 
I can 'pip install pyrsistent' just fine and that will install pyrsistent, but it fails when trying to install Mycodo.

kizniche, 2022-02-09:
Mycodo runs in a virtualenv, so you are not solving the issue by installing a pip package globally. The env is located at ~/Mycodo/env/

dylandn, 2022-02-09:
Which makes me wonder why it works globally but not in the virtualenv.

dylandn, 2022-02-09:
Part of the setup.log where it fails:
  Building wheel for pyrsistent (pyproject.toml): started
  Building wheel for pyrsistent (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error
  
  × Building wheel for pyrsistent (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [78 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-3.9
      copying _pyrsistent_version.py -> build/lib.linux-aarch64-3.9
      creating build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_field_common.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_pset.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_helpers.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_checked_types.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_pvector.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/typing.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_pmap.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_precord.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_pbag.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/__init__.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_immutable.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_pdeque.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_transformations.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_plist.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_toolz.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/_pclass.py -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/py.typed -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/__init__.pyi -> build/lib.linux-aarch64-3.9/pyrsistent
      copying pyrsistent/typing.pyi -> build/lib.linux-aarch64-3.9/pyrsistent
      running build_ext
      creating build/temp.linux-aarch64-3.9
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c pvectorcmodule.c -o build/temp.linux-aarch64-3.9/pvectorcmodule.o
      aarch64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-aarch64-3.9/pvectorcmodule.o -L/usr/lib -o build/lib.linux-aarch64-3.9/pvectorc.cpython-39-aarch64-linux-gnu.so
      running install
      running install_lib
      Traceback (most recent call last):
        File "/home/pi/Mycodo/env/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/home/pi/Mycodo/env/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pi/Mycodo/env/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 261, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 216, in build_wheel
          return self._build_with_temp_dir(['bdist_wheel'], '.whl',
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 202, in _build_with_temp_dir
          self.run_setup()
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 145, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 53, in <module>
          setup(
        File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 153, in setup
          return distutils.core.setup(**attrs)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 148, in setup
          dist.run_commands()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 967, in run_commands
          self.run_command(cmd)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 259, in run
          self.run_command('install')
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 61, in run
          return orig.install.run(self)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 578, in run
          self.run_command(cmd_name)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 985, in run_command
          cmd_obj.ensure_finalized()
[... cut ...]
          setattr(self, dst_option, getattr(src_cmd_obj, src_option))
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 103, in __getattr__
          raise AttributeError(attr)
      AttributeError: install_layout
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyrsistent
Failed to build pyrsistent
ERROR: Could not build wheels for pyrsistent, which is required to install pyproject.toml-based projects

kizniche, 2022-02-09:
Similar error recently at https://github.com/pypa/pip/issues/10857

kizniche, 2022-03-16:
I just committed a fix that I can confirm allows install on 64-bit Raspberry Pi OS. If you want to test, make sure you've deleted the current Mycodo directory, then get the latest code from master and install:

```bash
git clone https://github.com/kizniche/Mycodo
sudo ~/Mycodo/install/setup.sh
```

kizniche, 2022-03-16:
This issue has been mentioned on **DIY Automation Forum**. There might be relevant details there:

https://forum.kylegabriel.com/t/not-able-to-install-mycodo/782/3

closed 2022-03-28 by kizniche

LeoBosse, 2022-09-01:
I had a similar issue with an other packages, and it apparently comes from setuptools. Following this: [](https://github.com/pypa/setuptools/issues/3278)
I tried to run: `export SETUPTOOLS_USE_DISTUTILS=stdlib` in a terminal, and it worked like a charm!

kizniche, 2022-09-02:
What is the purpose? What issue does this resolve?

LeoBosse, 2022-11-08:
I found this thread while looking up my problem, and once I resolved it, I thought I would post my solution here to potentially help somebody with a similar error. I edited my comment with the link that didn't show for the source answer. If you feel that it is misplaced, feel free do remove it.

TheApeMachine, 2023-06-10:
> I had a similar issue with an other packages, and it apparently comes from setuptools. Following this: [pypa/setuptools#3278](https://github.com/pypa/setuptools/issues/3278) I tried to run: `export SETUPTOOLS_USE_DISTUTILS=stdlib` in a terminal, and it worked like a charm!

This worked perfectly to solve an error (AttributeError: install_layout. Did you mean: 'install_platlib'?) I was getting while building hnswlib as part of installing some python packages.

zsc, 2023-06-30:
In my case (not about Raspberry Pi but installing ffmpy), I got it work by "python3 -m pip install -U setuptools"