kizniche/Mycodo #1151: HC-SR04 Trigger pin and Echo pin configuration

mattimatti, 2022-02-07:
### Describe the problem/bug

While configuring a HC-SR04 sensor the custom options do not show because of the below linked line.
Not sure about how the "testing " mode cold be enabled.

https://github.com/kizniche/Mycodo/blob/798811d359b3a97f732774c06bd56210f74901a0/mycodo/inputs/hcsr04_circuitpython.py#L76

### Versions:

 - Mycodo Version: 8.12.9 
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: bullseye

### Reproducibility

1. Open configuration page for HC-SR04

### Expected behavior

Expected to fill the custom inputs Trigger pin and Echo pin in the gui

kizniche, 2022-02-07:
I don't understand the issue. I have no problem viewing or saving the two options you mention. Line 76 also has no affect on the display or saving of options. Can you explain the issue in greater detail (provide screenshots, etc.) and explain how you came to the conclusion line 76 is relevant?

mattimatti, 2022-02-08:
Attempting to configure the input, after installing the missing dependencies, the two controls did not render.

[image]

Since i was expecting them, went to the code and checked and found were defined in custom_options.

The only reference to custom_options is on lines 77-78-79 constrained by 76 so "if not testing:" is was the only apparent condition that could prevent the render of the controls.

Removed the line, adjusted indent, restarted the daemon, the controls rendered.

Before writing this reply I reverted my edit, restarted the daemon adn the inputs still display.

**is it relevant?**

Commenting that conditional allowed me to render and configure the controls as described above (**so was relevant**)
Uncommenting that conditional after configuration didn't hide the controls. (**so it's not relevant**)

If testing refers to the "unit" testing i shouldn't have been able to have that variable "true" in that and any moment.

I attach the Dependency installation log  just in case could help.

```
[2022-02-07 14:49:42] Dependency installation beginning. Installing: libgpiod-dev, adafruit_hcsr04

[2022-02-07 14:49:42] 
[2022-02-07 14:49:42] #### Installing/updating libgpiod-dev (apt)
[2022-02-07 14:49:43] Reading package lists...
[2022-02-07 14:49:44] Building dependency tree...
[2022-02-07 14:49:44] Reading state information...
[2022-02-07 14:49:47] The following package was automatically installed and is no longer required:
[2022-02-07 14:49:47]   libfuse2
[2022-02-07 14:49:47] Use 'apt autoremove' to remove it.
[2022-02-07 14:49:47] The following additional packages will be installed:
[2022-02-07 14:49:47]   libgpiod2
[2022-02-07 14:49:47] Suggested packages:
[2022-02-07 14:49:47]   libgpiod-doc
[2022-02-07 14:49:47] The following NEW packages will be installed:
[2022-02-07 14:49:47]   libgpiod-dev libgpiod2
[2022-02-07 14:49:48] 0 upgraded, 2 newly installed, 0 to remove and 62 not upgraded.
[2022-02-07 14:49:48] Need to get 87.5 kB of archives.
[2022-02-07 14:49:48] After this operation, 421 kB of additional disk space will be used.
[2022-02-07 14:49:48] Get:1 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libgpiod2 armhf 1.6.2-1+rpi1 [34.3 kB]
[2022-02-07 14:49:48] Get:2 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libgpiod-dev armhf 1.6.2-1+rpi1 [53.2 kB]
[2022-02-07 14:49:53] debconf: unable to initialize frontend: Dialog
[2022-02-07 14:49:53] debconf: (TERM is not set, so the dialog frontend is not usable.)
[2022-02-07 14:49:53] debconf: falling back to frontend: Readline
[2022-02-07 14:49:53] debconf: unable to initialize frontend: Readline
[2022-02-07 14:49:53] debconf: (This frontend requires a controlling tty.)
[2022-02-07 14:49:53] debconf: falling back to frontend: Teletype
[2022-02-07 14:49:53] dpkg-preconfigure: unable to re-open stdin: 
[2022-02-07 14:49:54] Fetched 87.5 kB in 1s (82.5 kB/s)
[2022-02-07 14:49:54] Selecting previously unselected package libgpiod2:armhf.
(Reading database ... 102083 files and directories currently installed.)
[2022-02-07 14:49:57] Preparing to unpack .../libgpiod2_1.6.2-1+rpi1_armhf.deb ...
[2022-02-07 14:49:57] Unpacking libgpiod2:armhf (1.6.2-1+rpi1) ...
[2022-02-07 14:49:57] Selecting previously unselected package libgpiod-dev:armhf.
[2022-02-07 14:49:57] Preparing to unpack .../libgpiod-dev_1.6.2-1+rpi1_armhf.deb ...
[2022-02-07 14:49:57] Unpacking libgpiod-dev:armhf (1.6.2-1+rpi1) ...
[2022-02-07 14:49:58] Setting up libgpiod2:armhf (1.6.2-1+rpi1) ...
[2022-02-07 14:49:58] Setting up libgpiod-dev:armhf (1.6.2-1+rpi1) ...
[2022-02-07 14:49:58] Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...

[2022-02-07 14:50:01] End install of libgpiod-dev

[2022-02-07 14:50:02] 
[2022-02-07 14:50:02] #### Installing/updating adafruit-circuitpython-hcsr04==0.4.6 (pip-pypi)
[2022-02-07 14:50:08] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2022-02-07 14:50:09] Collecting adafruit-circuitpython-hcsr04==0.4.6
[2022-02-07 14:50:09]   Downloading https://www.piwheels.org/simple/adafruit-circuitpython-hcsr04/adafruit_circuitpython_hcsr04-0.4.6-py3-none-any.whl (9.0 kB)
[2022-02-07 14:50:10] Requirement already satisfied: Adafruit-Blinka in ./env/lib/python3.9/site-packages (from adafruit-circuitpython-hcsr04==0.4.6) (6.18.0)
[2022-02-07 14:50:10] Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit-Blinka->adafruit-circuitpython-hcsr04==0.4.6) (0.7.0)
[2022-02-07 14:50:10] Requirement already satisfied: pyftdi>=0.40.0 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-hcsr04==0.4.6) (0.53.3)
[2022-02-07 14:50:10] Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-hcsr04==0.4.6) (1.1.9)
[2022-02-07 14:50:10] Requirement already satisfied: Adafruit-PlatformDetect>=3.13.0 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-hcsr04==0.4.6) (3.19.1)
[2022-02-07 14:50:10] Requirement already satisfied: rpi-ws281x>=4.0.0 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-hcsr04==0.4.6) (4.3.1)
[2022-02-07 14:50:10] Requirement already satisfied: sysv-ipc>=1.1.0 in ./env/lib/python3.9/site-packages (from Adafruit-Blinka->adafruit-circuitpython-hcsr04==0.4.6) (1.1.0)
[... cut ...]
[2022-02-07 14:50:20] WARNING: You are using pip version 21.3.1; however, version 22.0.3 is available.
[2022-02-07 14:50:20] You should consider upgrading via the '/home/pi/Mycodo/env/bin/python -m pip install --upgrade pip' command.

[2022-02-07 14:50:21] End install of adafruit_hcsr04

[2022-02-07 14:50:21] 
[2022-02-07 14:50:21] #### Setting permissions

[2022-02-07 14:50:25] #### Dependencies installed. Restarting frontend and backend...

[2022-02-07 14:50:39] #### Dependency install complete.
```

closed 2022-03-28 by kizniche