kizniche/Mycodo #1157: camera picture not attached and sent via e-mail

MariuszJac, 2022-03-04:
### Describe the problem/bug

### Versions:

Mycodo Version: 8.12.9
Python Version: 3.9.2 (default, Mar 12 2021, 04:06:34) [GCC 10.2.1 20210110]
Raspberry Pi Version: 4
Raspbian OS Version: Latest

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:
I've added function (conditional) that sends 2 e-mails: a) one is reporting sensor values, b) other is sending picture taken from raspi camera.

### Expected behavior

The first email is working fine, however, when I add another one as an action then it get's sent but does not include any attachment.
Name of the e-mail points to the correct directory where the camera file is located and I can see that it is there - however no attachment in the e-mail or file is sent. 
My e-mail box is gmail.
Camera configured as raspi and is working fine as well.
Tried setting up in separate trigger and still no luck - in this case no e-mail is sent at tall.
In debug mode looking at logs spotted that attachment type is set to 'none' but these are just logs.
If you could point me where in the code attachment sending is located I could debug it further.

### Screenshots
[image]

kizniche, 2022-03-04:
The Function Action system has recently had a major overhaul (https://github.com/kizniche/Mycodo/commit/4e2856128998cf3c783863f61e81a35574254be7), so I'll be testing the new system to ensure all actions work. It looks like the attachment path is being as the subject parameter rather than the attachment's.

MariuszJac, 2022-03-04:
Yes, I think that's what's happening - in logs I've seen sth like: attachment type: none so as you say - this could be the issue.
One other thing I've noticed is that for the latest raspi os version (bullseye) when adding CPU temp monitoring input I get this error: 
InputModule.get_measurement() method raised IOError: [Errno 2] No such file or directory: '/opt/vc/bin/vcgencmd'
It's caused by the fact that vcgencmd is in /usr/bin - making copy of it to the above pointing location sorts out the problem.
Sorry for mixing this in the comment but though you might want to know.

Overall it's a great piece of software that you've made - was initially aiming to build it using esp32 but you've convinced me to jump to raspi  :) Many thanks for sharing it as open source!

closed 2022-03-28 by kizniche