kizniche/Mycodo #1158: Measuring amount of water and electricity

garudaonekh, 2022-03-05:
Hi,
I need to measure the amount of water when I turn on a relay(Pump). I have a flow meter(pulse counter) and a pump(output). My question is how to measure the amount of water each time of turn on/off the pump. What complicated the matter is I have 5 pump connected with the same flow meter.

Any suggestion?

kizniche, 2022-03-05:
You can use the Generic Pump Output and calibrate them to the flow, then a volume and total volume is stored whenever the output is turned on/off.

garudaonekh, 2022-03-06:
But we are using flow sensor, so we need to bind pump and sensor

kizniche, 2022-03-06:
You would have to create an output to do that, however you will always run into an issue if you ever have more than one pump on at a time.

garudaonekh, 2022-03-06:
multiple pump but they are open sequentially. How about custom function? Can we use that instead of writing a new output?

kizniche, 2022-03-06:
no, an output is the mot appropriate.

garudaonekh, 2022-03-07:
Ok thanks.

closed 2022-03-07 by garudaonekh