kizniche/Mycodo #1159: Lost remote browser access

jacopienaar, 2022-03-12:
Not sure what happened but cannot access mycodo through the browser anymore and get  this error message:

Version: 8.12.9 
Database: 0187ea22dc4b 
Model: Raspberry Pi 3 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Dec 1 2021 15:08:00 
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start_x)

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1519, in full_dispatch_request
    return self.finalize_request(rv)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1540, in finalize_request
    response = self.process_response(response)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1888, in process_response
    self.session_interface.save_session(self, ctx.session, response)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_session/sessions.py", line 367, in save_session
    total_seconds(app.permanent_session_lifetime))
  File "/var/mycodo-root/env/lib/python3.7/site-packages/cachelib/file.py", line 147, in set
    self._prune()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/cachelib/file.py", line 87, in _prune
    if self._threshold == 0 or not self._file_count > self._threshold:
  File "/var/mycodo-root/env/lib/python3.7/site-packages/cachelib/file.py", line 58, in _file_count
    return self.get(self._fs_count_file) or 0
  File "/var/mycodo-root/env/lib/python3.7/site-packages/cachelib/file.py", line 127, in get
    return pickle.load(f)
EOFError: Ran out of input

I am running off master so kot sure if this has something to do with it

kizniche, 2022-03-16:
Ive never experienced that error, so not sure what the issue is.

jacopienaar, 2022-03-16:
It came right after a couple of restarts but it is a recurring issue (third time it has happened). Will see how it goes and do a clean install if it persists.

closed 2022-03-16 by jacopienaar