kizniche/Mycodo #1161: Can't have access to mycodo. Can't connect to localhost

Tommynator062, 2022-03-15:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.kylegabriel.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

I go to the terminal and i write curl -L https://kizniche.github.io/Mycodo/install | bash It start to install and then i get a message saying it can't connect to localhost (I joint the setup.log). I tried at home and at a different house and i still get the same issue

### Versions:

 - Mycodo Version: 8.12.9
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Buster

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. TERMINAL
2. curl -L https://kizniche.github.io/Mycodo/install | bash
3. error: 
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
4. Try to connect on the IP address but it refused the connection

### Expected behavior

Being able to log

Error from the setup.log:

#### Installing and configuring nginx web server
Removed /etc/systemd/system/multi-user.target.wants/mycodoflask.service.
Removed /etc/systemd/system/mycodoflask.service.
ln: failed to create symbolic link '/etc/nginx/sites-enabled/default': No such file or directory
Failed to enable unit: Unit file nginx.service does not exist.
Created symlink /etc/systemd/system/multi-user.target.wants/mycodoflask.service → /home/pi/Mycodo/install/mycodoflask.service.
Created symlink /etc/systemd/system/mycodoflask.service → /home/pi/Mycodo/install/mycodoflask.service.

#### Restarting nginx
Failed to restart nginx.service: Unit nginx.service not found.
#### Restarting mycodoflask

**#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...**

#### Setting permissions

#### Restarting the Mycodo daemon
#### Mycodo Installer finished Tue Mar 15 15:06:01 EDT 2022
#### Total install time: 16 minutes and 0 seconds

*********************************
** Mycodo finished installing! **
*********************************

Although the install finished, it doesn't necessarily mean it installed correctly.
If you experience issues, review the full install log located at:
/home/pi/Mycodo/install/setup.log

Go to https://132.207.115.45/, or whatever your device's
IP address is, to create an admin user and log in.

kizniche, 2022-03-16:
32 or 64-bit OS?

Fabesome, 2022-03-26:
Just my thoughts (beeing non-expert): requirements say "Raspberry Pi OS", you're stating "Raspbian OS Version: Buster".
Did you try to install the newest Raspberry Pi OS? Worked for me just fine with it yesterday, had problems on my old debian-os too.

closed 2022-03-28 by kizniche

Tree61, 2022-03-29:
Hi I have the same issue and have installed the latest ver of buster. The install setup log indicates err 404 in not founding nginx related files and failing to restart nginx.service. I am non expert appreciate how to fix this issue please.

[3-29-22 nginx error.pdf](https://github.com/kizniche/Mycodo/files/8368401/3-29-22.nginx.error.pdf)

kizniche, 2022-03-29:
Are you installing the latest Mycodo version, 8.13.1?

Tree61, 2022-03-30:
Thanks i think a bad connection via wifi better through enet 
completed os upgrade buster and mycodo ver 8.13.2
setup log looks clear of errors
now have connectivity to server