kizniche/Mycodo #1163: Timelapse - Not taking pictures after resuming past next capture date

jamjamjakjak, 2022-03-19:
Hello,

Many thanks to the founder(s) and to everyone who has contributed to this project. It is simply a piece of art!
I have a complicated setup and it is all functioning as intended.

I'm having an issue where Im trying to pause the timelapse during the dark period and resume in the morning.
It seems to get stuck and I think it may have something to do with the next capture date.
Its configured to take a picture every 2 hours(7200 seconds) however after many hours during the night paused, it is well past its 'Next Capture Date' and maybe waiting for this timestamp in the past to come again, causing it to fail to resume. 
Example:
[image]

Perhaps there is a way to programmically set the Next Capture Date manually?

Many thanks. hope this helps.

kizniche, 2022-03-24:
Thanks for the big report. I'll should have a fix shortly.

kizniche, 2022-03-24:
I don't see anything in the code that would prevent timelapses from working after a pause. Are you sure you're not out of disk space or something preventing photos form being captured? Any errors in the dameon log?

kizniche, 2022-03-24:
Your screenshot also indicates the daemon is not running, so that's likely your issue.

kizniche, 2022-03-25:
I was not able to replicate your issue. My time-lapse photo acquisition resumed following unpausing. You have some sort of issue separate from the time-lapse functionality.

kizniche, 2022-03-28:
If you have any additionla information, feel free to reopen

closed 2022-03-28 by kizniche

kizniche, 2022-06-15:
With some additional information on the forum, I was able to find and fix this issue with commit https://github.com/kizniche/Mycodo/commit/ed333330322202623b60c5b3fe25ade7a628dcf1.