kizniche/Mycodo #1166: user password lost after upgrade to version: 8.13.1 and Error 502: Bad Gateway Error

tezztuzer, 2022-03-29:
Hi, 

Thank you for mycodo! Great software!
I have two raspberry pi's running at different locations and wanted to upgrade them to the latest version 8.13.1. 
The previously installed mycodo version was v8.12.9
The upgrade completed successfully but after the upgrade, for both pi's the previously setup username/password combination didn't worked any more. 
I managed to reset the password via the file. 
After loging in, both rasbperry pi's returned the following error:
------
Error 502: Bad Gateway Error
Something bad happened but it's probably not your fault. Try refreshing the page to see if it will load. If that doesn't work, try rebooting your system, then try refreshing it again. If that still doesn't work, submit a [new issue on GitHub](https://github.com/kizniche/Mycodo/issues/new?template=problem-bug-report.md) with any useful information that can help determine the cause of this issue.

The last 50 lines from the Mycodo frontend web log can be viewed with the command:

journalctl -u mycodoflask | tail -n 50
If you create an issue on GitHub, provide this excerpt of the log for diagnostic purposes.
-------
One raspberry pi was rebooted and that solved the issue.
Unfortunately for the second raspberry pi reboot (multiple times) didn't helped. 

journalctl output:

journalctl -u mycodoflask | tail -n 50
-- Logs begin at Tue 2022-03-29 12:08:53 EEST, end at Tue 2022-03-29 12:12:51 EEST. --
Mar 29 12:09:01 salamior systemd[1]: Started Mycodo web user interface.
Mar 29 12:09:03 salamior python[382]: /var/mycodo-root/env/bin/python: No module named gunicorn
Mar 29 12:09:03 salamior systemd[1]: mycodoflask.service: Main process exited, code=exited, status=1/FAILURE
Mar 29 12:09:03 salamior systemd[1]: mycodoflask.service: Failed with result 'exit-code'.
 
I don't have physical, only remote access to the affected raspberry, 
Could this be fixed easily or better re-install mycodo? 

 - Mycodo Version: 8.13.1
 - both  Raspberry Pi Version: 2B+   (to old??)
 - Raspbian OS Version: 
  uname -a
Linux  myhost 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

Thank you.

kizniche, 2022-03-29:
Did you reboot, as instructed in the changelog?

kizniche, 2022-03-29:
Rebooting is instructed in the changelog to complete the upgrade. You do not need to reset the password, just reboot.

As for your second upgrade that was not successful, did you review the upgrade log?

kizniche, 2022-03-30:
How long did you give it to complete the upgrade before rebooting? You may have interrupted the upgrade.

tezztuzer, 2022-03-30:
Approximately 5 minutes I would say...Ok then I will re'install mycodo to get it back in working state. 
Thanks and have a great day!

kizniche, 2022-03-30:
You most likely interrupted the upgrade. 5 minutes is not a lot of time. You don't need to install. See if this command fixes whatever issue it has:

```
sudo ~/Mycodo/mycodo/scripts/upgrade_post.sh
```

tezztuzer, 2022-03-30:
That solved the issue.  
Thank you

closed 2022-03-30 by tezztuzer