kizniche/Mycodo #1167: rs485 support?

tomsepe, 2022-04-03:
I am looking at some industrial humidity/temperature sensors that use the Modbus protocol (RS485) such as this: https://www.vaisala.com/en/products/instruments-sensors-and-other-measurement-devices/instruments-industrial-measurements/hmd110-hmw110

I am wondering if there is a way to integrate Modbus sensors as a custom input?

I am considering interfacing them by using an RS485 shield https://www.hwhardsoft.de/english/projects/rs485-arduino/ and then using the Arduino to convert the data to i2c. 

What do you think about this approach?

Thank you!

kizniche, 2022-04-07:
>if there is a way to integrate Modbus sensors as a custom input?

I'm not sure. I've never worked with it, but it looks fairly straightforward. I'd be more inclined to try to interface directly to the Pi, rather than use an arduino, with only one intermediary IC, something like the following:

https://www.sparkfun.com/products/10124
https://www.sparkfun.com/products/9822
https://wiki.seeedstudio.com/RS-485_Shield_for_Raspberry_Pi/
https://www.amazon.com/RS485-CAN-HAT-Long-Distance-Communication/dp/B07VMB1ZKH

Josuah8, 2022-04-09:
Hi, you can use a cheap CH340/341 serial to USB converter stick and plug it into your pi, connect the modbus wires to it. In Mycodo just write a custom Python script to read the Modbus registers and return the values to Mycodo. Works like a charm. But integration like the other interface types would have saved me quite some time for sure.

lalebarde, 2022-04-12:
There are also cheap UART-RS485 converters like the [MAX485 module](https://medium.com/raspberry-pi-and-rs485-modbus/modbus-rs485-raspberry-pi-5ccbc1996b7d). I have just bought it, but I don't find out how to read sensors connected to the UART with mycodo. @[Josuah8](https://github.com/Josuah8), is your code applicable in my case, possibly changing the `tty`? Could you please elaborate on _"just write a custom Python script to read the Modbus registers and return the values to Mycodo"_?

Josuah8, 2022-04-12:
Hi, yes, if you manage to set the max485 or whatever other solution up
outside of mycodo then you know which tty to talk to. Then you can write a
script to read the values. Once this works you only have to adapt the
script to make it return values to mycodo as described in the manual
section about custom python scripts as input module. That's it. The python
script I posted works for my case, it uses minimalmodbus. The python
library you want to use must be installed in the virtual environment of
Mycodo, otherwise it won't find it. You should be able to modify my script
to match your modbus settings and the register actions you want to perform.
In my case I really got grey hair while trying to get it running outside of
mycodo. I had one usb stick with CH340 chip that was able to read data from
one sensor but not from another one and I was not able to figure out why.
Until I tried a different CH340 stick, it worked immediately. Even without
resistors and connecting the two lines to GND and voltage. The rest was
easy because the script was already correct. I needed some testing with the
data types to get meaningful data out of the sensor and also figured out
that the documentation of the sensor was a bit misleading concerning the
register addresses. But I got it up and running and it's working well since
a year already. If we had a modbus interface in mycodo it would make it a
lot easier to change settings on the fly without having to edit the python
script.

hth
Josuah

kizniche, 2022-05-10:
This issue has been mentioned on **Radical DIY Forum**. There might be relevant details there:

https://forum.radicaldiy.com/t/custom-input-for-industrial-ec-with-rs485-protocol/751/2

friedpenguin, 2022-07-13:
Haha. You pointed the forum back to here. MODBUS would be a cool addition because a lot of solar charge controllers use MODBUS as well. And the way you've integrated MQTT to send data to/from where needed is awesome. Great project! Getting some traction on Reddit too.

kizniche, 2023-06-06:
This issue has been mentioned on **Radical DIY Forum**. There might be relevant details there:

https://forum.radicaldiy.com/t/brainstorming-par-photosynthetically-active-radiation-light-sensor-for-mycodo/1557/1

tazomatalax, 2024-03-04:
I want to connect a modbus pH probe and have the following code:
```
`from pymodbus.client.sync import ModbusSerialClient
import struct

# Create a Modbus client object
client = ModbusSerialClient(method='rtu', port='ttyUSB0', baudrate=19200, bytesize=8, parity='N', stopbits=2, timeout=1)

# Connect to the Modbus device
client.connect()

# Read 10 registers starting from address 2090
ph_values = client.read_holding_registers(address=2090, count=10, unit=1)

# Read 10 registers starting from address 2410
temp_values = client.read_holding_registers(address=2410, count=10, unit=1)

# Close the Modbus connection
client.close()

# Convert the register values to bytestrings
ph_bytes = bytearray(ph_values.registers[2:4])
temp_bytes = bytearray(temp_values.registers[2:4])

# Interpret the bytestrings as measurements
pH = struct.unpack('f', ph_bytes)[0]
temperature = struct.unpack('f', temp_bytes)[0]

self.store_measurement(channel=0, measurement=pH)
self.store_measurement(channel=1, measurement=temperature)

# # Print the measurements
# print(f"pH: {pH}")
# print(f"Temperature: {temperature}")
```

and have installed pymodbus with `sudo ~/Mycodo/env/bin/pip install pymodbus` 

Confirmed is to be installed with 

```
pi@fac-reterminal:~ $ sudo ~/Mycodo/env/bin/pip list
Package                Version
---------------------- --------
alembic                1.11.2
aniso8601              9.0.1
astroid                3.0.3
attrs                  23.2.0
Babel                  2.14.0
bcrypt                 3.2.2
beautifulsoup4         4.12.3
blinker                1.7.0
Brotli                 1.1.0
cachelib               0.12.0
certifi                2024.2.2
cffi                   1.16.0
charset-normalizer     3.3.2
ciso8601               2.3.1
click                  8.1.7
decorator              5.1.1
Deprecated             1.2.14
dill                   0.3.8
distro                 1.8.0
dnspython              2.6.1
email-validator        1.3.1
factory-boy            3.2.1
Faker                  23.2.1
filelock               3.12.2
Flask                  2.3.2
flask-accept           0.0.6
[... 82 lines cut ...]
pi/Mycodo/mycodo/user_python_code/input_python_code_effc8924-a189-49fe-8c35-b6cbe3579804.py:13:4: R0913: Too many arguments (6/5) (too-many-arguments)
pi/Mycodo/mycodo/user_python_code/input_python_code_effc8924-a189-49fe-8c35-b6cbe3579804.py:50:8: W0612: Unused variable 'message' (unused-variable)
pi/Mycodo/mycodo/user_python_code/input_python_code_effc8924-a189-49fe-8c35-b6cbe3579804.py:55:8: E0401: Unable to import 'pymodbus.client.sync' (import-error)
pi/Mycodo/mycodo/user_python_code/input_python_code_effc8924-a189-49fe-8c35-b6cbe3579804.py:55:8: C0415: Import outside toplevel (pymodbus.client.sync.ModbusSerialClient) (import-outside-toplevel)
pi/Mycodo/mycodo/user_python_code/input_python_code_effc8924-a189-49fe-8c35-b6cbe3579804.py:55:8: E0611: No name 'sync' in module 'pymodbus.client' (no-name-in-module)
pi/Mycodo/mycodo/user_python_code/input_python_code_effc8924-a189-49fe-8c35-b6cbe3579804.py:56:8: C0415: Import outside toplevel (struct) (import-outside-toplevel)

------------------------------------------------------------------
Your code has been rated at 6.74/10 (previous run: 6.74/10, +0.00)

```

in the check and:

```
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/python_code.py", line 288, in get_measurement
    run.python_code_run()
  File "/home/pi/Mycodo/mycodo/user_python_code/input_python_code_effc8924-a189-49fe-8c35-b6cbe3579804.py", line 55, in python_code_run
    from pymodbus.client.sync import ModbusSerialClient
ModuleNotFoundError: No module named 'pymodbus.client.sync'
2024-03-04 09:21:34,683 - DEBUG - mycodo.controllers.controller_input_effc8924 - Adding measurements to InfluxDB with ID effc8924-a189-49fe-8c35-b6cbe3579804: {}
````

in the logs. I have tried restarting frontend and backend, then rebooting the pi. Still nothing works. What am i doing wrong?