kizniche/Mycodo #1168: Failed Update from 8.13.2 to 8.13.4 - Mycodo not working

donplumpos, 2022-04-04:
Pi: Raspberry PI 4 B 8GB
OS: Bullseye
Initial Mycodo version: 8.13.2
Target Mycodo version. 8.13.4

Initiated update via frontend. After several minutes i tried to reload page but get "Error 502: Bad Gateway Error"

Last lines in mycodo.log:
`2022-04-04 12:32:06,732 - ERROR - mycodo.modules - Could not load module: Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/modules.py", line 19, in load_module_from_file
    spec.loader.exec_module(module_custom)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/plumps/Mycodo/mycodo/widgets/widget_graph_synchronous.py", line 41, in <module>
    from mycodo.mycodo_flask.utils.utils_general import use_unit_generate
  File "/var/mycodo-root/mycodo/mycodo_flask/utils/utils_general.py", line 15, in <module>
    from importlib_metadata import version
ModuleNotFoundError: No module named 'importlib_metadata'

2022-04-04 12:32:06,776 - ERROR - mycodo.modules - Could not load module: Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/modules.py", line 19, in load_module_from_file
    spec.loader.exec_module(module_custom)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/plumps/Mycodo/mycodo/widgets/widget_graph_synchronous.py", line 41, in <module>
    from mycodo.mycodo_flask.utils.utils_general import use_unit_generate
  File "/var/mycodo-root/mycodo/mycodo_flask/utils/utils_general.py", line 15, in <module>
    from importlib_metadata import version
ModuleNotFoundError: No module named 'importlib_metadata'

2022-04-04 12:32:06,806 - ERROR - mycodo.modules - Could not load module: Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/modules.py", line 19, in load_module_from_file
    spec.loader.exec_module(module_custom)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/plumps/Mycodo/mycodo/widgets/widget_graph_synchronous.py", line 41, in <module>
    from mycodo.mycodo_flask.utils.utils_general import use_unit_generate
  File "/var/mycodo-root/mycodo/mycodo_flask/utils/utils_general.py", line 15, in <module>
    from importlib_metadata import version
ModuleNotFoundError: No module named 'importlib_metadata'

2022-04-04 12:32:06,835 - ERROR - mycodo.modules - Could not load module: Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/modules.py", line 19, in load_module_from_file
    spec.loader.exec_module(module_custom)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/plumps/Mycodo/mycodo/widgets/widget_graph_synchronous.py", line 41, in <module>
    from mycodo.mycodo_flask.utils.utils_general import use_unit_generate
  File "/var/mycodo-root/mycodo/mycodo_flask/utils/utils_general.py", line 15, in <module>
    from importlib_metadata import version
ModuleNotFoundError: No module named 'importlib_metadata'

2022-04-04 12:32:06,857 - ERROR - mycodo.controllers.controller_widget - 'widget_graph_synchronous' is not a valid widget type
2022-04-04 12:32:06,869 - ERROR - mycodo.modules - Could not load module: Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/utils/modules.py", line 19, in load_module_from_file
    spec.loader.exec_module(module_custom)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/plumps/Mycodo/mycodo/widgets/widget_graph_synchronous.py", line 41, in <module>
    from mycodo.mycodo_flask.utils.utils_general import use_unit_generate
  File "/var/mycodo-root/mycodo/mycodo_flask/utils/utils_general.py", line 15, in <module>
    from importlib_metadata import version
ModuleNotFoundError: No module named 'importlib_metadata'

2022-04-04 12:32:06,890 - ERROR - mycodo.controllers.controller_widget - 'widget_graph_synchronous' is not a valid widget type
2022-04-04 12:32:06,890 - ERROR - mycodo.controllers.controller_widget - 'widget_graph_synchronous' is not a valid widget type
2022-04-04 12:32:06,890 - ERROR - mycodo.controllers.controller_widget - 'widget_graph_synchronous' is not a valid widget type
2022-04-04 12:32:06,891 - INFO - mycodo.controllers.controller_widget - Activated in 204.4 ms
2022-04-04 12:32:07,892 - INFO - mycodo.daemon - Mycodo daemon started in 4.267 seconds
2022-04-04 12:32:07,894 - INFO - mycodo.daemon - 56.29 MB RAM in use`
[... 26 lines cut ...]
Apr 04 12:31:27 plumpsrpi python[584]:   File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
Apr 04 12:31:27 plumpsrpi python[584]:   File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
Apr 04 12:31:27 plumpsrpi python[584]:   File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
Apr 04 12:31:27 plumpsrpi python[584]:   File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
Apr 04 12:31:27 plumpsrpi python[584]:   File "<frozen importlib._bootstrap_external>", line 790, in exec_module
Apr 04 12:31:27 plumpsrpi python[584]:   File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
Apr 04 12:31:27 plumpsrpi python[584]:   File "/home/plumps/Mycodo/mycodo/start_flask_ui.py", line 11, in <module>
Apr 04 12:31:27 plumpsrpi python[584]:     from mycodo.mycodo_flask.app import create_app
Apr 04 12:31:27 plumpsrpi python[584]:   File "/home/plumps/Mycodo/mycodo/mycodo_flask/app.py", line 28, in <module>
Apr 04 12:31:27 plumpsrpi python[584]:     from mycodo.mycodo_flask import routes_admin
Apr 04 12:31:27 plumpsrpi python[584]:   File "/home/plumps/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 47, in <module>
Apr 04 12:31:27 plumpsrpi python[584]:     from mycodo.mycodo_flask.routes_static import inject_variables
Apr 04 12:31:27 plumpsrpi python[584]:   File "/home/plumps/Mycodo/mycodo/mycodo_flask/routes_static.py", line 28, in <module>
Apr 04 12:31:27 plumpsrpi python[584]:     from mycodo.mycodo_flask.forms import forms_dashboard
Apr 04 12:31:27 plumpsrpi python[584]:   File "/home/plumps/Mycodo/mycodo/mycodo_flask/forms/forms_dashboard.py", line 19, in <module>
Apr 04 12:31:27 plumpsrpi python[584]:     from mycodo.mycodo_flask.utils.utils_general import generate_form_widget_list
Apr 04 12:31:27 plumpsrpi python[584]:   File "/home/plumps/Mycodo/mycodo/mycodo_flask/utils/utils_general.py", line 15, in <module>
Apr 04 12:31:27 plumpsrpi python[584]:     from importlib_metadata import version
Apr 04 12:31:27 plumpsrpi python[584]: ModuleNotFoundError: No module named 'importlib_metadata'
Apr 04 12:31:27 plumpsrpi python[584]: [2022-04-04 12:31:27 +0200] [584] [INFO] Worker exiting (pid: 584)
Apr 04 12:31:28 plumpsrpi python[499]: [2022-04-04 12:31:28 +0200] [499] [INFO] Shutting down: Master
Apr 04 12:31:28 plumpsrpi python[499]: [2022-04-04 12:31:28 +0200] [499] [INFO] Reason: Worker failed to boot.
Apr 04 12:31:28 plumpsrpi systemd[1]: mycodoflask.service: Main process exited, code=exited, status=3/NOTIMPLEMENTED
Apr 04 12:31:28 plumpsrpi systemd[1]: mycodoflask.service: Failed with result 'exit-code'.
Apr 04 12:31:28 plumpsrpi systemd[1]: mycodoflask.service: Consumed 7.176s CPU time.`

kizniche, 2022-04-04:
What version python?

donplumpos, 2022-04-04:
Python 3.9.2

kizniche, 2022-04-04:
And is that the version of python in the virtualenv at ~/Mycodo/env?

donplumpos, 2022-04-04:
Yes. 

~/Mycodo/env/pyvenv.cfg:

home = /usr
implementation = CPython
**version_info = 3.9.2.final.0**
virtualenv = 20.7.0
include-system-site-packages = false
base-prefix = /usr
base-exec-prefix = /usr
base-executable = /usr/bin/python3

kizniche, 2022-04-04:
Thanks for the bug report. I was able to replicate the issue and just released v8.13.5. You can upgrade from the command line with the command:

```bash
sudo ~/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade-mycodo
```

donplumpos, 2022-04-04:
Update to 8.13.5 was successfully, frontend is working again. Thank you for your fast response and your great work.

closed 2022-04-04 by donplumpos