kizniche/Mycodo #1171: Trigger Edge timer not working

Tree61, 2022-04-08:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

*[A clear and concise description of what you were trying to do and what the problem/bug is - replace this with your own text]*

Using GPIO 2 switched input via external pull up resistor arrangement
to edge trigger output GPIO 3 externally wired to an output relay
edge trigger duration tested with 0, 5 and 10sec duration
output just raises and falls when observed with a voltage meter on GPIO 3 ie no time on observed 
live data confirms GPIO 2 state change however the edge trigger input does not change state
forcing relay out also works
changed de bounce settings 50-150msec same result

### Versions:

 - Mycodo Version: *[e.g. 8.0.3]* latest
 - Raspberry Pi Version: *[e.g. 3B+]* 4
 - Raspbian OS Version: *[e.g. Buster Lite]* buster

### Reproducibility yes repeatable result many times

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *[Go to...]* refer above
2. *[Click on...]*
3. *[Scroll down to...]*
4. *[See error...]*
5. *[replace this list with your own text]*

### Expected behavior - Should observe relay output staying high for set duration period

*[A clear and concise description of what you expected to happen - replace this with your own text]*

### Screenshots

*[If applicable, add screenshots to help explain your problem - replace this with your own text]*

### Additional context

*[Anything that should be added to make it easier to address this issue - replace this with your own text]*

Tree61, 2022-04-09:
Also tried configuring trigger timer function no success with this either
screen shots of configuration attache
[image]
[image]
[image]
[image]
[image]
[image]
[image]
d

kizniche, 2022-04-16:
I found an issue with the Edge Input storing measurements which I just committed a fix for, but I didn't find an issue with the Edge Trigger. I tested as you have done it worked for me.

closed 2022-04-22 by kizniche