kizniche/Mycodo #1172: Python Module not working inside Mycodo

TravisDuck999, 2022-04-08:
First, just want to say thanks for all the amazing work, I am really enjoying using this software busy setting up a system to test if it all works as I hope now.

So my problem is when I run a Python3 code input.
When I run the same code in command line it works great, but when I put it in Mycodo I have this problem.
( From the logs )

`2022-04-08 15:40:11,373 - ERROR - mycodo.inputs.python_code_f5b57a57 - 1
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/inputs/python_code.py", line 234, in get_measurement
    run.python_code_run()
  File "/home/pi/Mycodo/mycodo/user_python_code/input_python_code_f5b57a57-7ad0-47f8-92fb-585791895f2e.py", line 33, in python_code_run
    from uFire_pH import uFire_pH# import the needed PH lib
  File "/home/pi/PH_code/Isolated_ISE/python/RaspberryPi/uFire_pH.py", line 1, in <module>
    from uFire_ISE import uFire_ISE
  File "/home/pi/PH_code/Isolated_ISE/python/RaspberryPi/uFire_ISE.py", line 4, in <module>
    import smbus  # pylint: disable=E0401
ModuleNotFoundError: No module named 'smbus'`

I know smbus is installed as the whole thing works on Command line.

The library I am using is a custom one and I need to use :
`sys.path.append("/home/pi/PH_code/Isolated_ISE/python/RaspberryPi/")`

In the Python3 code to get the library avaliable to be used by Mycodo, but it seems "smbus" in there is not happy and can't be used.

I had this working fine before I upgraded, from about 3.12.9 >> 3.13.7

Any help would be great.

kizniche, 2022-04-08:
I just updated the docs for the Python Code Input to include information about the Python environment. Here is the update to the docs. Please let me know if it's not clear or can be improved in any way.

> Your code is executed from the same Python virtual environment that Mycodo runs from. Therefore, you must install Python libraries to this environment if you want them to be available to your code. This virtualenv is located at ~/Mycodo/env and if you wanted to install a library, for example "my_library" using pip, you would execute "sudo ~/Mycodo/env/bin/pip install my_library".

TravisDuck999, 2022-04-09:
You are amazing, I should have read the Docs but I got complacent with my knowledge from the last revisions, thanks for the help.

closed 2022-04-09 by TravisDuck999

kizniche, 2022-04-09:
The above quote was what I added to the docs in response to this issue you opened. Prior to this issue, it was unclear how to install python modules that could be used within the Python input. So, thank you for bringing the oversight to my addention, because it helped strengthen the docs.