kizniche/Mycodo #1173: Raspberry Pi 4 issues

kurtisa727, 2022-04-10:
I am using the system set up by Kyle on his YouTube video on automated hydroponics. Its a great system with a lot of thought and development involved. I'm preparing to film a Youtube video using most of Kyle's Set Up combined with my own irrigation system so I definitely want to work through this and get this system online, but I'm stumped. 

I'll give you the list of bugs and then attach the log entries. 

I'm running a Raspberry Pi 4 with the 32 bit Raspberry Pi operating system and version 8.13.7 of Mycodo and version 3.9.2 of Python. I have actually reloaded both Mycodo and the operating system and reproduced the same issues set forth below. 

I set my system up pretty much like the original video and accompanying information provided by Kyle. (Again, lots of thought went into this set up on your end). I actually disassembled my system and just back tracked all my connections to no avail. My wire length is very short to reduce any noise or power issues. I don't see any issues with my connections or hardware set up.
, 
I'm using: (7) Atlas Scientific EZO-Pumps, Atlas Scientific Water Temp Sensor, Atlas Scientific PH Sensor, Atlas Scientific Electrical Conductivity Sensor, 20x4 LCD Screen, Digital to Analog Converter, HTU 21 Sensor, Kasa HS300 6 outlet Wi-fi plug. (I am not using the water level set up or the flow meter and accompanying chip. 

I am getting readings from the HTU 21 Sensor, the Digital to Analog Converter, and both the PH Sensor and EC Sensor HOWEVER, the system won't permit me to calibrate the PH or EC sensors. I am not getting a reading from the temp sensor at all. 

When I plug and EZO-Pumps into the system, as I plug them in, they show up at different addresses. Pumps 1-5 ended up with different addresses as I plugged them in, pumps 6 and 7 were tagged with the same address as pump 5. The system would not permit me to change ANY of the addresses of any of the pumps. Pumps 2-5 will dispense their calibration amounts, Pump 1 will not. I tried changing the addresses as I was adding pumps and after all the pumps are plugged in. No difference. 

The 20x4 LCD is blank and once in a while it displays gibberish. I'm a nociveHere's 300 lines of log entry so you get an idea what's going on...

2022-04-10 00:07:30,663 - DEBUG - mycodo.inputs.atlas_pt1000_6f74f320 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:07:30,664 - ERROR - mycodo.inputs.atlas_pt1000_6f74f320 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:07:32,865 - DEBUG - mycodo.devices.atlas_scientific_i2c_102_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:07:32,866 - DEBUG - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:07:32,866 - ERROR - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:07:45,588 - DEBUG - mycodo.devices.atlas_scientific_i2c_65_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:07:45,588 - DEBUG - mycodo.inputs.atlas_pt1000_6f74f320 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:07:45,589 - ERROR - mycodo.inputs.atlas_pt1000_6f74f320 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:07:45,589 - ERROR - mycodo.controllers.controller_input_6f74f320 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-04-10 00:07:47,891 - DEBUG - mycodo.devices.atlas_scientific_i2c_102_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:07:47,892 - DEBUG - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:07:47,892 - ERROR - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:00,622 - DEBUG - mycodo.devices.atlas_scientific_i2c_65_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:08:00,622 - DEBUG - mycodo.inputs.atlas_pt1000_6f74f320 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:08:00,623 - ERROR - mycodo.inputs.atlas_pt1000_6f74f320 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:02,825 - DEBUG - mycodo.devices.atlas_scientific_i2c_102_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:08:02,825 - DEBUG - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:08:02,826 - ERROR - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:02,826 - ERROR - mycodo.controllers.controller_input_4d3e4cb4 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-04-10 00:08:15,647 - DEBUG - mycodo.devices.atlas_scientific_i2c_65_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:08:15,647 - DEBUG - mycodo.inputs.atlas_pt1000_6f74f320 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:08:15,648 - ERROR - mycodo.inputs.atlas_pt1000_6f74f320 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:17,850 - DEBUG - mycodo.devices.atlas_scientific_i2c_102_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:08:17,851 - DEBUG - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:08:17,851 - ERROR - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:30,673 - DEBUG - mycodo.devices.atlas_scientific_i2c_65_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:08:30,673 - DEBUG - mycodo.inputs.atlas_pt1000_6f74f320 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:08:30,674 - ERROR - mycodo.inputs.atlas_pt1000_6f74f320 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:30,674 - ERROR - mycodo.controllers.controller_input_6f74f320 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-04-10 00:08:32,876 - DEBUG - mycodo.devices.atlas_scientific_i2c_102_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:08:32,877 - DEBUG - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:08:32,877 - ERROR - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:45,605 - DEBUG - mycodo.devices.atlas_scientific_i2c_65_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:08:45,606 - DEBUG - mycodo.inputs.atlas_pt1000_6f74f320 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:08:45,606 - ERROR - mycodo.inputs.atlas_pt1000_6f74f320 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:47,911 - DEBUG - mycodo.devices.atlas_scientific_i2c_102_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:08:47,912 - DEBUG - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:08:47,912 - ERROR - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:08:47,912 - ERROR - mycodo.controllers.controller_input_4d3e4cb4 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-04-10 00:09:00,633 - DEBUG - mycodo.devices.atlas_scientific_i2c_65_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:09:00,634 - DEBUG - mycodo.inputs.atlas_pt1000_6f74f320 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:09:00,634 - ERROR - mycodo.inputs.atlas_pt1000_6f74f320 - Sensor read unsuccessful: [Errno 121] Remote I/O error
2022-04-10 00:09:02,839 - DEBUG - mycodo.devices.atlas_scientific_i2c_102_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
[... cut ...]
2022-04-10 00:18:47,179 - DEBUG - mycodo.controllers.controller_input_6f74f320 - Adding measurements to InfluxDB with ID 6f74f320-8ed2-48d8-89e2-4daa18d9fdf7: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 0.0, 'timestamp_utc': None}}
2022-04-10 00:18:47,869 - DEBUG - mycodo.devices.atlas_scientific_i2c_102_1 - AtlasScientificI2C raised an exception when taking a reading: [Errno 121] Remote I/O error
2022-04-10 00:18:47,869 - DEBUG - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Device Returned: error: [Errno 121] Remote I/O error
2022-04-10 00:18:47,869 - ERROR - mycodo.inputs.atlas_pt1000_4d3e4cb4 - Sensor read unsuccessful: [Errno 121] Remote I/O error

kurtisa727, 2022-04-10:
I saw that the LCD screen is no longer supported in Mycodo

kizniche, 2022-04-11:
I asked you to please use the forum for this discussion, sine it is likely a configuration issue, not an issue with the software. https://github.com/kizniche/Mycodo-custom/issues/5#issuecomment-1092335971

closed 2022-04-11 by kizniche