kizniche/Mycodo #1180: Can't Open Dependencies Page From The Menu

dcgris, 2022-04-21:
Hi,
I'm new to the community, so I may not have done the best job with this but here goes.

### Describe the problem/bug:
After updating to the newest version of Mycodo (8.13.9), I tried to navigate to the dependencies page from the menu. After clicking the link in the menu, the page loads for a considerable amount of time, to eventually bring me to an Error 500 (Internal Server Error) page.
Here's the entire traceback below:

> Error (Full Traceback):
> 
> Traceback (most recent call last):
>   File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2077, in wsgi_app
>     response = self.full_dispatch_request()
>   File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1525, in full_dispatch_request
>     rv = self.handle_user_exception(e)
>   File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 672, in error_router
>     return original_handler(e)
>   File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1523, in full_dispatch_request
>     rv = self.dispatch_request()
>   File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1509, in dispatch_request
>     return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
>   File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 277, in decorated_view
>     return current_app.ensure_sync(func)(*args, **kwargs)
>   File "/home/srprototype/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 383, in admin_dependencies
>     if each_dep not in unmet_list:
> TypeError: unhashable type: 'list'

### Versions:
_Version: 8.13.9
Database: b354722c9b8b
Model: Raspberry Pi 4 Model B Rev 1.4

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

Firmware:
b''_

### Reproducibility 
1) From any page, click the settings menu icon (gear icon) in the top right corner of the page.
2) Select the dependencies link from the drop down menu.

### Expected Behaviour
When clicking on the dependencies link, the dependencies page should appear.

**I'm open to any feedback on my bug reporting.
 Thanks, have a good day folks.**

kizniche, 2022-04-21:
Thanks for the bug report. What Inputs, Outputs, and Functions do you have installed?

snickers2k, 2022-04-22:
same here

having installed anyleaf and not much more (almost fresh install) .. uninstalled everything (settings, depencies, upgrade file)
but still the same

also i noticed the raspberry is completely unreliable .. crashes every few minutes sometimes
clicking on "logview" results in crash aswell

and also i don't get measurements from anyleaf (of course probably)
```

 Version: 8.13.9
Database: b354722c9b8b
Model: Raspberry Pi 3 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

Firmware:
b''

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 383, in admin_dependencies
    if each_dep not in unmet_list:
TypeError: unhashable type: 'list'
```

the install was a pain in the ass aswell. it took me like 15 tries (with unplugging the raspberry from power) to get mycodo installed, because server problems and so on.

dcgris, 2022-04-22:
> Thanks for the bug report. What Inputs, Outputs, and Functions do you have installed?

I had just started setting it up, I only have the CPU temp configured as an input. Nothing else.

kizniche, 2022-04-22:
>[2022-04-22 17:30:35] sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: camera.json_headers

not related. this is an issue with your system (#1181)

kizniche, 2022-04-23:
I installed Mycodo on Debian on an Orange Pi 4 this evening and was able to replicate the issue on a fresh install, and committed a fix that will be available in the next release.

dcgris, 2022-04-27:
Awesome, thanks Kyle!

closed 2022-05-20 by kizniche