kizniche/Mycodo #1183: Fail to upgrade to 8.13.8

pe2269, 2022-04-26:
Hi,

I want to upgrade to the 8.13.8 version of Mycodo but after reboot i got this error.

Thanks a lot for all this amazing work on Mycodo !!

Version: 8.13.8
Database: b354722c9b8b
Model: Raspberry Pi 4 Model B Rev 1.4

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

Firmware:
Dec 1 2021 15:01:54
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1820, in _execute_context
    cursor, statement, parameters, context
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 732, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: camera.show_preview

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_dashboard.py", line 104, in page_dashboard
    camera = Camera.query.all()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 2768, in all
    return self._iter().all()
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 2906, in _iter
    execution_options={"_sa_orm_load_options": self.load_options},
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/orm/session.py", line 1692, in execute
    result = conn._execute_20(statement, params or {}, execution_options)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1631, in _execute_20
    return meth(self, args_10style, kwargs_10style, execution_options)
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 326, in _execute_on_connection
    self, multiparams, params, execution_options
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1508, in _execute_clauseelement
    cache_hit=cache_hit,
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1863, in _execute_context
    e, statement, parameters, cursor, context
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 2044, in _handle_dbapi_exception
    sqlalchemy_exception, with_traceback=exc_info[2], from_=e
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 207, in raise_
    raise exception
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1820, in _execute_context
    cursor, statement, parameters, context
  File "/var/mycodo-root/env/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 732, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: camera.show_preview
[SQL: SELECT camera.id AS camera_id, camera.unique_id AS camera_unique_id, camera.name AS camera_name, camera.library AS camera_library, camera.device AS camera_device, camera.opencv_device AS camera_opencv_device, camera.hflip AS camera_hflip, camera.vflip AS camera_vflip, camera.rotation AS camera_rotation, camera.brightness AS camera_brightness, camera.contrast AS camera_contrast, camera.exposure AS camera_exposure, camera.gain AS camera_gain, camera.hue AS camera_hue, camera.saturation AS camera_saturation, camera.white_balance AS camera_white_balance, camera.custom_options AS camera_custom_options, camera.output_id AS camera_output_id, camera.output_duration AS camera_output_duration, camera.cmd_pre_camera AS camera_cmd_pre_camera, camera.cmd_post_camera AS camera_cmd_post_camera, camera.stream_started AS camera_stream_started, camera.hide_still AS camera_hide_still, camera.hide_timelapse AS camera_hide_timelapse, camera.url_still AS camera_url_still, camera.url_stream AS camera_url_stream, camera.show_preview AS camera_show_preview, camera.output_format AS camera_output_format, camera.timelapse_started AS camera_timelapse_started, camera.timelapse_paused AS camera_timelapse_paused, camera.timelapse_start_time AS camera_timelapse_start_time, camera.timelapse_end_time AS camera_timelapse_end_time, camera.timelapse_interval AS camera_timelapse_interval, camera.timelapse_next_capture AS camera_timelapse_next_capture, camera.timelapse_capture_number AS camera_timelapse_capture_number, camera.timelapse_last_file AS camera_timelapse_last_file, camera.timelapse_last_ts AS camera_timelapse_last_ts, camera.still_last_file AS camera_still_last_file, camera.still_last_ts AS camera_still_last_ts, camera.path_still AS camera_path_still, camera.path_timelapse AS camera_path_timelapse, camera.path_video AS camera_path_video, camera.width AS camera_width, camera.height AS camera_height, camera.resolution_stream_width AS camera_resolution_stream_width, camera.resolution_stream_height AS camera_resolution_stream_height, camera.stream_fps AS camera_stream_fps, camera.picamera_shutter_speed AS camera_picamera_shutter_speed, camera.picamera_sharpness AS camera_picamera_sharpness, camera.picamera_iso AS camera_picamera_iso, camera.picamera_awb AS camera_picamera_awb, camera.picamera_awb_gain_red AS camera_picamera_awb_gain_red, camera.picamera_awb_gain_blue AS camera_picamera_awb_gain_blue, camera.picamera_exposure_mode AS camera_picamera_exposure_mode, camera.picamera_meter_mode AS camera_picamera_meter_mode, camera.picamera_image_effect AS camera_picamera_image_effect 
FROM camera]
(Background on this error at: https://sqlalche.me/e/14/e3q8)

kizniche, 2022-04-30:
Your upgrade failed. Try to [restore a backup](https://github.com/kizniche/Mycodo/wiki/Backup-and-Restore#restore-a-full-mycodo-backup).

pe2269, 2022-05-02:
Yes it's working now :) 
Thanks !!