kizniche/Mycodo #1188: DFRobot Digital Peristaltic Pump - DFR0523

snickers2k, 2022-05-08:
Altough i have no experience with this pump, i would love to have some alternatives for the atlas scientific digital pumps.
It's not too cheap, with 2/3 of the price of the atlas ones, so $65 instead of $95 - but if they are good like them, could be a great alternative.

https://wiki.dfrobot.com/Gravity__Digital_Peristaltic_Pump_SKU__DFR0523

thank you

kizniche, 2022-05-12:
I don't see I2C listed for the product you linked to.

snickers2k, 2022-05-13:
you are right
so without i2c, i guess this one is problematic with multiple pumps? maybe i figure out another alternative for atlas pumps :/

closed 2022-05-13 by snickers2k

kizniche, 2022-05-13:
Not necessarily. It will just require a GPIO to drive a PPM signal for each pump, which could also potentially be controlled by a 16-channel I2C PWM driver.

kizniche, 2022-05-13:
I bought this pump along with the temperature board/probe, so I will be seeing if I can make an output module for it.

snickers2k, 2023-09-04:
any news on this?

kizniche, 2023-09-05:
I had an issue when I initially started developing the output module and I haven't made a second attempt. I'll give it another go and see if I can get a working output made.

kizniche, 2023-09-05:
>can somebody explain why the Atlas are worth the price

Also, to answer this question, the pumps themselves are high quality Welco pumps. I have several of these pumps by themselves (~$30 each) that I control with MOSFETs. The rest of the Atlas Scientific circuitry on the pump is good for beginners that are trying to interface it with different communication protocols and want to only send a volume or duration to turn on rather than coding all aspects of the driver. With Mycodo, this becomes redundant, since there are now general pump outputs that can conduct calibration and dispense volumes automatically based on the calibrated flow rate. These outputs didn't exist when I created my first howto that used these pumps. I created them later as a way to use less expensive pumps and your own control circuitry (relay, MOSFET, etc.) to reduce the cost.

kizniche, 2023-09-05:
The AS pump circuit also adds a bit of reliability, especially when dispensing for long durations, as you can send a duration to the pump and its own microcontroller will schedule turning the pump off. If relying on Mycodo to turn the pump on and then off after a duration, and something happened to Mycodo (or the hardware running it) before the pump turned off and never gets the off signal, the pump may stay running beyond when it should have turned off. Having dedicated hardware to handle turning the pump off improves reliability.

snickers2k, 2023-09-05:
this is well explained. thank you very much. that level of reliability AS-pumps has seems a good point to take into consideration.

kizniche, 2023-09-05:
I'll add that the current Mycodo Atlas Scientific Pump Output module does ***not*** send the duration to the pump to handle turning off automatically. The module operates as all the other duration modules do, which is Mycodo handles turning the pump on and off. To take advantage of the AS Pump's ability to manage its own timer-handling, a new output module would need to be created (simply duplicate the current Atlas Scientific Pump Output and make necessary changes). I haven't had a need for such an output, since my pump durations are relatively short (1 - 3 seconds at a time), but as on duration increases, so too does the risk (when not using the built-in timer-handling).