kizniche/Mycodo #1189: DFRobot PT1000 Temperature-Sensor - DFR0558

snickers2k, 2022-05-12:
As an alternative I2C-Method for the Atlas PT1000 Temperature Sensors, these ones from DFRobot only costs $25
would be great to have them as official input

https://www.dfrobot.com/product-1753.html

thanks

kizniche, 2022-05-12:
However, I just noticed only UART is supported by the current MAX31855 library, whereas the DFRobot library uses I2C. I'll see about adding the additional library to support the DFRobot I2C device.

kizniche, 2022-05-12:
I just added this Input in https://github.com/kizniche/Mycodo/commit/936421452645c78a367c88c86b1f6bdf32f8598a

I don't have this, but it should work, if the dfrobot library works. I probably buy this and can verify once it arrives.

snickers2k, 2022-05-12:
amazing. thank you!

closed 2022-05-20 by kizniche