kizniche/Mycodo #1190: fresh install failed

smit9418, 2022-05-13:
fresh OS "2022-04-04-raspios-bullseye-arm64.img"
update and upgrade ran
after curl -L https://kizniche.github.io/Mycodo/install | bash
i get this with no progress

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1293  100  1293    0     0   4992      0 --:--:-- --:--:-- --:--:--  4992
Get:1 http://security.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Err:1 http://security.debian.org/debian-security bullseye-security InRelease
  Error writing to file - write (28: No space left on device) [IP: 2a04:4e42:5f::644 80]
Hit:2 http://deb.debian.org/debian bullseye InRelease
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Err:3 http://deb.debian.org/debian bullseye-updates InRelease
  Error writing to file - write (28: No space left on device) [IP: 2a04:4e42:5f::644 80]
Err:2 http://deb.debian.org/debian bullseye InRelease
  Splitting up /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye_InRelease into data and signature failed
Get:4 http://archive.raspberrypi.org/debian bullseye InRelease [23.7 kB]
Err:4 http://archive.raspberrypi.org/debian bullseye InRelease
  Error writing to file - write (28: No space left on device) [IP: 2a00:1098:84:1e0::1 80]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://deb.debian.org/debian bullseye InRelease: Splitting up /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye_InRelease into data and signature failed
W: Failed to fetch http://deb.debian.org/debian/dists/bullseye/InRelease  Splitting up /var/lib/apt/lists/deb.debian.org_debian_dists_bullseye_InRelease into data and signature failed
W: Failed to fetch http://security.debian.org/debian-security/dists/bullseye-security/InRelease  Error writing to file - write (28: No space left on device) [IP: 2a04:4e42:5f::644 80]
W: Failed to fetch http://deb.debian.org/debian/dists/bullseye-updates/InRelease  Error writing to file - write (28: No space left on device) [IP: 2a04:4e42:5f::644 80]
W: Failed to fetch http://archive.raspberrypi.org/debian/dists/bullseye/InRelease  Error writing to file - write (28: No space left on device) [IP: 2a00:1098:84:1e0::1 80]
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
whiptail is already the newest version (0.52.21-4+b3).
python3 is already the newest version (3.9.2-3).
python3 set to manually installed.
wget is already the newest version (1.21-1+deb11u1).
wget set to manually installed.
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  libjq1 libonig5
The following NEW packages will be installed:
  jq libjq1 libonig5
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 361 kB of archives.
After this operation, 1,117 kB of additional disk space will be used.
E: You don't have enough free space in /var/cache/apt/archives/.
bash: line 32: jq: command not found
No URLs found in -.
(23) Failed writing body
mkdir: cannot create directory ‘Mycodo’: No space left on device

gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now
bash: line 36: cd: Mycodo/install: No such file or directory
/bin/bash: ./setup.sh: No such file or directory

kizniche, 2022-05-13:
> Error writing to file - write (28: No space left on device)

Looks like your disk is full

smit9418, 2022-05-13:
> > Error writing to file - write (28: No space left on device)
> 
> Looks like your disk is full

doesnt make any sense, 256gb sd card with nothing else on it

kizniche, 2022-05-13:
The size of the card doesn't necessarily mean that's the size of your partition(s) or free space. What does the `df -h` command yield?

smit9418, 2022-05-16:
> The size of the card doesn't necessarily mean that's the size of your partition(s) or free space. What does the `df -h` command yield?

i got a chance to revisit it tonight, and due to my lack of recent rpi usage i forgot to expand the filesystem after installing to OS. i got the system up and running. thank you for the reply and pointing me in the right direction. just subbed to your youtube channel and im stoked to get into this.

closed 2022-05-16 by smit9418

kizniche, 2022-05-16:
Strange... Raspberry Pi OS has automatically expanded the filesystem upon the first boot for over a year now.

smit9418, 2022-05-16:
> Strange... Raspberry Pi OS has automatically expanded the filesystem upon the first boot for over a year now.

Idk I just remembered having to do it manually so I hit up menuconfig and it allowed me to install after that

kizniche, 2022-05-16:
Maybe there was an issue with the automated process at the first boot.