kizniche/Mycodo #1194: Conditional Statement Pylint Warning

mdschoff, 2022-05-18:
After editing my pH and EC variables in the conditional statement and saving them, I keep getting a warning that Pylint is returning these messages. Is this something that will affect the automation of dosage to control the pH and EC values? If so how do I go about fixing this issue?

[image]

kizniche, 2022-05-18:
It's a warning, and the pylint return status is displayed at the top of the Conditional. Warnings are not the same as Errors, since they are of lesser severity. You need to view the pylint output to determine if the code you saved is valid or not.

closed 2022-05-18 by kizniche

kizniche, 2022-05-18:
It sounds like you're not running the latest version of Mycodo. If you are running the latest version, please attach a screenshot of the missing Conditional Status and the entire top of the config page, following a successful save.

kizniche, 2022-05-18:
I see both the pylint output that appears at the top of the Conditional Controller config page as well as the Conditional Status field. I'm not sure what you mean by them not being there.

mdschoff, 2022-05-18:
Sorry, I just didn't know if there was supposed to be a value within the conditional status part and I am unsure what the pylint output is within the code.

kizniche, 2022-05-18:
There is a default value in the Status text field, but it looks like you deleted it.
The pylint output shows the entire generated class and the pylint score/issues at the bottom.

mdschoff, 2022-05-18:
I see. I don't remember where I got the conditional code. Is there a master code I can copy and redo my variable assignment? 

Sorry for all the questions. This conditional statement I think is the only thing between me and the completed system.

kizniche, 2022-05-18:
This is the pH/EC Conditional setup that you should use, if you're not familiar with building your own code: https://kylegabriel.com/projects/2020/06/automated-hydroponic-system-build.html#Electrical_Conductivity_and_pH_Regulation