kizniche/Mycodo #1196: Add K-96 multi gas GHG sensor for CH4, CO2, N2O

alonrab, 2022-05-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Dear Mycodo team, 
We are developing a new open-source soil greenhouse gas reader for farm application. this will use the newer Sense-air K-96 NDIR sensor or the existing K30 NDIR sensor to record CH4 and CO2 gas flux. This project is part of a USDA NIFA-supported program for open-source farming.

I'm interested in a new feature - adding a K-96 sensor.
 
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I need to add the K96 code to the input library. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

This will supplement the K30 sensor option with a newer generation of more precise multi-gas NDIR sensors.

**Additional context**
Add any other context or screenshots about the feature request here.

Sens-air provided the python code they are using to operate the sensor - see attached.

[K96_ReadLog_v220516.txt](https://github.com/kizniche/Mycodo/files/8740610/K96_ReadLog_v220516.txt)

kizniche, 2022-06-14:
It's not immediately clear to me, looking at the code, where the final measurement values are located. I don't have the sensor, so I can't actually run the code. Can you provide some more information?

alonrab, 2022-08-09:
I got the sensor, now trying to run it. seem like the data is being saved to a folder. It is a UART connection 
I attached the product install file [
[K96_Integration (003).pdf](https://github.com/kizniche/Mycodo/files/9292681/K96_Integration.003.pdf)
](url)

kizniche, 2022-11-06:
Any progress with this? Can I be of any assistance?

alonrab, 2022-11-08:
I got the sensor to run with python code, and it works ok. The sensor
software saves data as a CSV file, and I would like your help in importing
it into the MyCodo database
I attached the python code, output file, and explanation of the output
file. NaN is no-reading saved. I think that once a air sampler dryer is
added, it will give accurate methane readings
Many thanks!

kizniche, 2022-11-09:
You need to use a web browser if you want to add files to a comment.

alonrab, 2022-11-09:
got it; attached files 

[k96 sensor code for pi.txt](https://github.com/kizniche/Mycodo/files/9966570/k96.sensor.code.for.pi.txt)
[reader k96 explained.xlsx](https://github.com/kizniche/Mycodo/files/9966571/reader.k96.explained.xlsx)
[k96221024_RawData.csv](https://github.com/kizniche/Mycodo/files/9966572/k96221024_RawData.csv)

kizniche, 2022-11-09:
I just added the K96. You can upgrade to master to test.

kizniche, 2022-11-16:
If that's the case, it may be best to provide them with a system that's already set up. Mycodo has become mostly hardware agnostic, as long as a Debian-based OS is installed.
Do you have a link where this sensor can be purchased?

alonrab, 2022-11-16:
yes, this is the plan; I am making a download ready *img file. The K96 is still in the prototype stage at sensair. I'm hoping it will be available soon, they sell the K30 in the U.S so hopefully, once they are happy with the K96 it will be available too.
thanks,

closed 2024-10-04 by kizniche