kizniche/Mycodo #1197: Change I2C address for Catnip Chirp moisture sensor

alonrab, 2022-05-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Dear MyCodo team,
I'm trying to install 10 catnip chirp soil moisture readers using I2C and would like to change the hex address from the standard 0x20 and do the calibration. Can the change address and calibration codes be added to the MyCodo input?
I have 20 of these, and they are currently being used for greenhouse irrigation control, soil gas reading, soil radon reading, and many other applications. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Add the change address option to the sensor input, so multiple devices can be used simultaneously.

Add the moisture calibration option
 
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

N/A
**Additional context**
Add any other context or screenshots about the feature request here.

I think the python code provided is in python 2, and this seems to be an issue with the current raspberry os

https://github.com/ageir/chirp-rpi

#!/usr/bin/python

#https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/tree/master/Adafruit_I2C
from Adafruit_I2C import Adafruit_I2C
from time import sleep, strftime
from datetime import datetime
deviceAddr = 0x20

i2c = Adafruit_I2C( deviceAddr, -1, False )

#to change adress
#i2c.write8( 1, 0x22 )

#reset sensor, we need this otherwise i get inconsistent light reading in the dark...
i2c.write8( deviceAddr, 0x06 )
sleep(5)

i2c.write8(deviceAddr, 3)
sleep(3)
light = i2c.readU16(4, False)
temp = i2c.readS16(5, False)/float(10)
moisture = i2c.readU16(0, False)
print "Temperature\tMoisture\tBrightness"
print str(temp) + ":" + str(moisture) + ":" + str(light)

Many thanks,

kizniche, 2022-05-23:
The code I was using for the input already had the ability to set the I2C address. I just added the ability to use the function from the input config. However, it's untested, so if you would like to test and report back, that would be appreciated.

alonrab, 2022-05-24:
I tried this today but could not change it from the default 0x20 address. I did not see the change address option? I updated the latest version and tried to tweak the input configuration - no change in the i2c hex address. 
Thanks,

alonrab, 2022-06-04:
Was able to change the I2C address after update to master version, thank you much
Another question/request for consideration: calibrate each reading of temp, moisture, light
this will have the user take a reading of soil temp with chirp then put corresponding temp probe reading value (one point calibration) that will convert the value from chirp in the database. Similarly, chirp take reading from known soil moisture of dry, saturated, 50% soil samples and put input (3 point calibration) that will convert the value of soil moisture reading in the database. Similar action for light reading and put light value (one point calibration).
Right now I'm converting data after extraction to excel which is a big hindrance for field applications.  This will be very useful for water conservative irrigation 
Many thanks,