kizniche/Mycodo #1198: Feature Request: Calibration Completion Confirmation

mdschoff, 2022-05-23:
There seems to be no feedback accessible from the input page where you calibrate the instruments when calibrating the instruments. This means I don't know exactly when the instruments are done with their calibration process. 

If it is possible to receive a popup just like when the calibration button is pressed initially to see that the system has sent a calibration complete message would be great to ensure that my devices are fully calibrated to the best of my abilities. 

It is possible that within the logs there is some sort of confirmation, however, It is a hassle to go from inputs to log and search through the log to find such information. I also realize that it is possible that there isn't a clear signal from the Atlas instruments saying that they are calibrated. Another solution could be built into the inputs page where you calibrate with the button turning green once the calibration is complete. 

I realize that this is sort of insignificant in the physical function of mycodo, however, I think this feature would benefit the program and its users. 

If somehow I missed something or some sort of confirmation of calibration completion I haven't noticed please let me know. 
Thank you for your work on this project and continued support of the project.

kizniche, 2022-05-23:
>It is possible that within the logs there is some sort of confirmation

It states right on the Input config page where the calibration occurs, "Status messages will be set to the Daemon Log, accessible from Config -> Mycodo Logs -> Daemon Log"

>It is a hassle to go from inputs to log and search through the log to find such information.

How is it a hassle? It's only 3 clicks.

mdschoff, 2022-05-24:
Thank you for your reply!

I am sorry if this feature request isn't something you see fit to be implemented. Just thought that if I thought it would be beneficial that others might find it the same. 

I realize that it is just 3 clicks, but again having it all in one place would make it a little more efficient for the user to make the process smoother. 

Again this was just my opinion and I would hope that even if nothing comes of it that this is a fine place to post about feature requests when users find things to be added or improved upon.

kizniche, 2022-06-01:
I will see about being more verbose to the user on the Input Page when calibrating.

closed 2023-01-14 by kizniche