kizniche/Mycodo #1200: Some Report i should upload.

ser6iogl92, 2022-05-30:
Describe the problem/bug:

When i click on functions, or outputs, appear the ''Error 500.: Internal Server Error.

Versions:

Version: 8.13.10
Model: Raspberry Pi 3 Model B Rev 1.2 
OS: Raspberry Pi OS Lite 32bits

In the main page, I click on Functions or Outputs, and appear the Error 500: Internal server error.

Expected behavior:
I expeted to be open like normally do. I was managing some outputs some minutes before and was working good, but after a while, it dont open anymore, i rebooted the device

Screenshoots:
I can not add images then i will put all the text like in the page:
Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a [new issue on GitHub](https://github.com/kizniche/Mycodo/issues/new?template=problem-bug-report.md) with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.13.10
Database: 743de2cd05e3
Model: Raspberry Pi 3 Model B Rev 1.2

Release:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

Firmware:
b''

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1819, in _execute_context
    self.dialect.do_execute(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/engine/default.py", line 732, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: camera.json_headers

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_function.py", line 367, in page_function
    camera = Camera.query.all()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/orm/query.py", line 2768, in all
    return self._iter().all()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/orm/query.py", line 2903, in _iter
    result = self.session.execute(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1692, in execute
    result = conn._execute_20(statement, params or {}, execution_options)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1631, in _execute_20
    return meth(self, args_10style, kwargs_10style, execution_options)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/sql/elements.py", line 325, in _execute_on_connection
    return connection._execute_clauseelement(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1498, in _execute_clauseelement
    ret = self._execute_context(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1862, in _execute_context
    self._handle_dbapi_exception(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 2043, in _handle_dbapi_exception
    util.raise_(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/util/compat.py", line 207, in raise_
    raise exception
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1819, in _execute_context
    self.dialect.do_execute(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sqlalchemy/engine/default.py", line 732, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: camera.json_headers
[SQL: SELECT camera.id AS camera_id, camera.unique_id AS camera_unique_id, camera.name AS camera_name, camera.library AS camera_library, camera.device AS camera_device, camera.opencv_device AS camera_opencv_device, camera.hflip AS camera_hflip, camera.vflip AS camera_vflip, camera.rotation AS camera_rotation, camera.brightness AS camera_brightness, camera.contrast AS camera_contrast, camera.exposure AS camera_exposure, camera.gain AS camera_gain, camera.hue AS camera_hue, camera.saturation AS camera_saturation, camera.white_balance AS camera_white_balance, camera.custom_options AS camera_custom_options, camera.output_id AS camera_output_id, camera.output_duration AS camera_output_duration, camera.cmd_pre_camera AS camera_cmd_pre_camera, camera.cmd_post_camera AS camera_cmd_post_camera, camera.stream_started AS camera_stream_started, camera.hide_still AS camera_hide_still, camera.hide_timelapse AS camera_hide_timelapse, camera.url_still AS camera_url_still, camera.url_stream AS camera_url_stream, camera.json_headers AS camera_json_headers, camera.show_preview AS camera_show_preview, camera.output_format AS camera_output_format, camera.timelapse_started AS camera_timelapse_started, camera.timelapse_paused AS camera_timelapse_paused, camera.timelapse_start_time AS camera_timelapse_start_time, camera.timelapse_end_time AS camera_timelapse_end_time, camera.timelapse_interval AS camera_timelapse_interval, camera.timelapse_next_capture AS camera_timelapse_next_capture, camera.timelapse_capture_number AS camera_timelapse_capture_number, camera.timelapse_last_file AS camera_timelapse_last_file, camera.timelapse_last_ts AS camera_timelapse_last_ts, camera.still_last_file AS camera_still_last_file, camera.still_last_ts AS camera_still_last_ts, camera.path_still AS camera_path_still, camera.path_timelapse AS camera_path_timelapse, camera.path_video AS camera_path_video, camera.width AS camera_width, camera.height AS camera_height, camera.resolution_stream_width AS camera_resolution_stream_width, camera.resolution_stream_height AS camera_resolution_stream_height, camera.stream_fps AS camera_stream_fps, camera.picamera_shutter_speed AS camera_picamera_shutter_speed, camera.picamera_sharpness AS camera_picamera_sharpness, camera.picamera_iso AS camera_picamera_iso, camera.picamera_awb AS camera_picamera_awb, camera.picamera_awb_gain_red AS camera_picamera_awb_gain_red, camera.picamera_awb_gain_blue AS camera_picamera_awb_gain_blue, camera.picamera_exposure_mode AS camera_picamera_exposure_mode, camera.picamera_meter_mode AS camera_picamera_meter_mode, camera.picamera_image_effect AS camera_picamera_image_effect 
FROM camera]
(Background on this error at: https://sqlalche.me/e/14/e3q8)

Additional context

I dont have nothing more to add. only thanks to the developers

kizniche, 2022-05-30:
Your database didn't upgrade properly. Try running this:

`cd ~/Mycodo/alembic_db && sudo ../env/bin/alembic upgrade head`

ser6iogl92, 2022-05-30:
yes, is all working again. thanks again

closed 2022-06-01 by kizniche