kizniche/Mycodo #1202: Raspberry pico start issue and solution

jamjamjakjak, 2022-06-16:
Hello, thanks for fixing the time lapse issue, I forgot to submit my fix. Your notification did remind me of another.

The raspberry pico after a few influxdb entries hangs on boot, you need to increase the timeout within the influxdb startup script. The pico can run this project but needs more time for it's tiny processing power to boot. The frontend comes up, just a DB timing out and restarting loop.

All my units are in production away from me, next time I'll check for the timeout adjustments. It required two edits of variables.

Hope this helps. I use many picos these days as pi is bit expensive with shortages.

Cheers

kizniche, 2022-06-16:
Do you mean the Pi Zero? The Pico is a microcontroller.

jamjamjakjak, 2022-06-16:
I'm terribly sorry, yes. I'm currently overwhelmed with sickness

Raspberry Pi Zero, please update title or I'll try when I wake up. Zz

Thanks and sorry for the inconvenience

jamjamjakjak, 2022-07-15:
Sorry about the delay.

I finally got into the Zero unit and it seems I hackily modified:
nano /usr/lib/influxdb/scripts/influxd-systemd-start.sh

and changed
 echo "InfluxDB API unavailable after $attempts attempts..."
  sleep 1

to

 echo "InfluxDB API unavailable after $attempts attempts..."
  sleep 3

Ultimately extending the timeout period to allow its tiny processor to boot the influxdb.

Only other issues is that when processing Function > Conditional Controller you must wait a few minutes for the process to complete, it main see it has hanged, but it is processing and needs its time.
Otherwise is very functional.

Hope this helps other users looking to use this unit.

closed 2022-10-10 by kizniche