kizniche/Mycodo #1203: Add an import field in the conditional controler

lalebarde, 2022-06-16:
**Is your feature request related to a problem? Please describe.**
I need to use functions from python libraries in the conditional controller Conditional Statement part. If I put the `import` here, it will be executed at each call. Even if Python is optimised, that can seriously affect performance in some circumstances. [PEP 8](http://www.python.org/dev/peps/pep-0008/#imports) states "Imports are always put at the top of the file".

[image]

**Describe the solution you'd like**
Add a field before the "Conditional Statement" one to enable adding imports that would be put at the top of the module.
[image]

kizniche, 2022-06-17:
I like that idea, it's a fairly easy change to implement but adds a lot of additional value to the Conditional Function. Thanks for the feedback.

lalebarde, 2022-06-18:
Great. To maximaze useability, it should be in fact a general "start module" section, where one could add anything than  imports.

lalebarde, 2022-06-21:
I make a lot of errors with IDs. This "start module" will enable me also to declare an action function where all IDs will be written once:

def raw_action(id, message, log=message):
    self.run_action("{" + id + "}", message=message)
    self.logger.info(log)

def action(sAction):
    if sAction == "ON EV watering":
        raw_action("a4477a33", "ON EV watering")
    elif sAction == .....

kizniche, 2022-08-06:
I just added support for both import and class initialization code blocks. You can [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) if you want to test the latest code prior to release.

kizniche, 2022-08-13:
This pull request has been mentioned on **Radical DIY Forum**. There might be relevant details there:

https://forum.radicaldiy.com/t/mycodo-v8-14-0-release/1136/1

closed 2022-08-13 by kizniche

lalebarde, 2023-05-27:
Hi Kyle,
Would it be possible please to be able to increase the edition window for the _Initialize Python Code_ frame, and of course all of them, possibly a full screen mode?
As it is, I have to copy the content and edit it in a text editor since only 4 lines are shown.