kizniche/Mycodo #1204: Improve visual separation between inputs/outputs in conditional controlers to avoid confusion

lalebarde, 2022-06-19:
After a first error of mine mixing output IDs with wrong output like here in red:

[image]

I still have difficulties to read correctly the outputs like I should in green, the wrong red reading is more "natural" to me.

It would be nice to improve I/O visual separation in the layout.

Mondgans, 2022-06-21:
I can confirm this. 
For Inputs the ID is below the dropdown field. And for the Outputs the ID is above the dropdown field.
This is definitely confusing.

kizniche, 2022-07-03:
Fixed in https://github.com/kizniche/Mycodo/commit/60fb718b78aa5a0a880a0ae21d47cd633055cc65

closed 2022-11-06 by kizniche