kizniche/Mycodo #1207: PID is not controlling output

BearBrew, 2022-07-03:
New install to control a mini-fridge for beer fermentation purposes.  Have a DHT22 temperature sensor that MyCodo is able to read and keep history.  Using Raspberry Pi GPIO pins 23/24 to control a solid state relay, model by SainSmart.  One input and 2 outputs placed in MyCodo and MyCodo is able to MANUALLY toggle the output relays.  I placed a PID with a set point, however it never automatically toggles the output even when the setpoint is reached.  Looking at the logs, I have this:

2022-07-03 14:40:46,478 - DEBUG - mycodo.controllers.controller_pid_415c28f1 - Latest (CH0, Unit: F): 66.74 @ 2022-07-03 14:40:36
2022-07-03 14:40:46,479 - DEBUG - mycodo.controllers.controller_pid_415c28f1 - PID: Input: 66.74, Output: P: -6.239999999999995, I: -0.0, D: 0.0, Out: -6.239999999999995
2022-07-03 14:41:07,013 - DEBUG - mycodo.controllers.controller_input_d649448b - Adding measurements to InfluxDB with ID d649448b-dfb0-448c-a218-8f18d36d662b: {0: {'measurement': None, 'unit': 'F', 'value': 66.74, 'timestamp_utc': datetime.datetime(2022, 7, 3, 18, 41, 6, 974163)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 99.4, 'timestamp_utc': datetime.datetime(2022, 7, 3, 18, 41, 6, 974185)}}
2022-07-03 14:41:16,456 - DEBUG - mycodo.controllers.controller_pid_415c28f1 - Latest (CH0, Unit: F): 66.74 @ 2022-07-03 14:41:07
2022-07-03 14:41:16,457 - DEBUG - mycodo.controllers.controller_pid_415c28f1 - PID: Input: 66.74, Output: P: -6.239999999999995, I: -0.0, D: 0.0, Out: -6.239999999999995

Looks to me like it reads the input and compares the values but just never does anything with it.  What am I missing?

 - Mycodo Version: 8.13.10
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: Bullseye Lite

kizniche, 2022-07-03:
It's likely user error/misconfiguration.

kizniche, 2022-07-03:
A PID is also not recommended for fridge/freezer control unless you really know what you're doing with a PID or you'll likely damage your frisge/freezer. A Bang-bang Function is the preferred method.

BearBrew, 2022-07-03:
OK,  I will try that then.   What do you use on the dashboard?  I mean how do you set the setpoint 

⁣Get BlueMail for Android ​

kizniche, 2022-10-10:
There isn't currently a widget to change a Bang-Bang setpoint, so you'll have to do that from the Function page (or create a custom Widget to do that and add a helper function in the Bang-Bang Function to receive that command and change the setpoint).

closed 2022-11-01 by kizniche