kizniche/Mycodo #1215: ADC MCP3208

mattyhunter, 2022-07-17:
Hey mate, i want to ask you, if there is any possibility to add support for ADC MCP3208. It should be very similar to MCP 3008,.

Thank you for your reply. Have a nice day!
M.

kizniche, 2022-08-10:
I can't find a Python library for this device.

kizniche, 2022-08-10:
I found a library that is no longer in pypi, at https://github.com/MomsFriendlyRobotCompany/mcp3208. I'm unsure if it works and I don't have this device to test. Nonetheless, I created an Input module that uses the library and should work, in commit https://github.com/kizniche/Mycodo/commit/4a1bcf89c03fe80d3d260aa22253ff111a582d45. To test the latest this module in the latest code, prior to release, you'll need to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch).

kizniche, 2022-08-13:
This pull request has been mentioned on **Radical DIY Forum**. There might be relevant details there:

https://forum.radicaldiy.com/t/mycodo-v8-14-0-release/1136/1

closed 2023-01-14 by kizniche