kizniche/Mycodo #1222: Turkish Language Support Add

ramazansancar, 2022-08-29:
**Is your feature request related to a problem? Please describe.**
I can do the Turkish translation for you.

**Describe the solution you'd like**
[I can do the Turkish translation for you.](http://translate.kylegabriel.com:8080/projects/mycodo/)
Language can be added for translation.

**Describe alternatives you've considered**

**Additional context**

kizniche, 2022-08-29:
Hi. That is great. You can find information on how to create a new translation file at https://github.com/kizniche/Mycodo/wiki/Translations

ramazansancar, 2022-08-31:
#1223 added :)

kizniche, 2022-09-07:
Thanks! I'll see about adding it.

kizniche, 2022-09-07:
I just committed the translation file, so it now appears in [weblate](http://translate.kylegabriel.com:8080/projects/mycodo/master/), currently only 7% complete, Weblate makes it quick and easy to update translations, if you wanted to increase the coverage.

[image]

closed 2022-11-22 by kizniche