kizniche/Mycodo #1232: Exception while reading measurement from the influxdb database

sagehenstudio, 2022-10-07:
### Describe the problem/bug

Outputs are not running because values cannot be retrieved from database?

### Versions:

 - Mycodo Version: 8.14.2
 - Raspberry Pi Version: Zero 2 W
 - Raspbian OS Version: Debian GNU/Linux 11 (bullseye)

### Reproducibility

Everything seemed to be working very well before I shut down the system in March. Here in October I didn't have success re-starting it (same errors as posted below) so assumed some sort of DB mismatch issue since when I left I was still in Mycodo v.8.12.9. So I wiped the Pi and loaded the most recent version of Mycodo fresh. Honestly I cannot understand how this is not happening to other people, but also, knowing "me," it's probably something really niche.

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Follow instructions at (https://kylegabriel.com/projects/2021/09/mushroom-cultivation-automation.html#Install_Raspberry_Pi_OS) to set up Pi, inputs, outputs and functions. Inputs are currently an Atlas Humidity sensor on i2c and a MH-Z19B, both are reading measurements successfully on the "Live Measurements" screen.
2. The outputs, currently including a fan and humidifier, _are not running at all_. Again, these are set up per the instructions at kylegabriel.com.
3. Errors in the log look like this:
`2022-10-07 09:57:55,132 - INFO - mycodo.devices.atlas_scientific_i2c_1_111 - Atlas Scientific Board: HUM, Rev: 2, Firmware: 1.0
2022-10-07 09:57:57,174 - INFO - mycodo.controllers.controller_input_4514a10b - Activated in 3257.9 ms
2022-10-07 09:57:57,828 - DEBUG - mycodo.controllers.controller_input_74fe2bec - Pre output successfully set up
2022-10-07 09:57:57,986 - INFO - mycodo.controllers.controller_input_74fe2bec - Activated in 577.6 ms
2022-10-07 09:57:57,987 - INFO - mycodo.daemon - All activated Input controllers started
2022-10-07 09:57:59,203 - INFO - mycodo.controllers.controller_pid_9be1dfa4 - PID Settings: Device ID: 4514a10b-7731-45ca-b577-02aaf7bbd478, Measurement ID: 46571f7e-8545-452b-a71a-6811108c8589, Direction: raise, Period: 120.0, Setpoint: 97.0, Band: 0.0, Kp: 3.0, Ki: 0.25, Kd: 0.0, Integrator Min: -100.0, Integrator Max 100.0, Output Raise: afef748c-8b91-4241-a167-709058add954, Output Raise Channel: 2, Output Raise Type: on_off, Output Raise Min On: 10.0, Output Raise Max On: 0.0, Output Raise Min Off: 20.0, Output Raise Always Min: False, Output Lower: None, Output Lower Channel: None, Output Lower Type: None, Output Lower Min On: 0.0, Output Lower Max On: 0.0, Output Lower Min Off: 0.0, Output Lower Always Min: False, Setpoint Tracking Type: , Setpoint Tracking ID: 
2022-10-07 09:57:59,204 - INFO - mycodo.controllers.controller_pid_9be1dfa4 - Activated in 530.4 ms
2022-10-07 09:57:59,204 - INFO - mycodo.daemon - All activated PID controllers started
2022-10-07 09:57:59,212 - INFO - mycodo.daemon - All activated Function controllers started
2022-10-07 09:57:59,500 - INFO - mycodo.controllers.controller_widget - Activated in 286.8 ms
2022-10-07 09:58:00,504 - INFO - mycodo.daemon - Mycodo daemon started in 10.728 seconds
2022-10-07 09:58:00,586 - INFO - mycodo.daemon - 91.03 MB RAM in use
2022-10-07 09:58:03,394 - ERROR - mycodo.controllers.controller_pid_9be1dfa4 - Exception while reading measurement from the influxdb database
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 451, in get_last_measurement_pid
    utc_dt = datetime.datetime.strptime(self.last_time.split(".")[0], '%Y-%m-%dT%H:%M:%S')
AttributeError: 'float' object has no attribute 'split'`
4. I confirmed that in /var/mycodo-root/mycodo/controllers/controller_pid.py, self.last_time is a float.
5. I have tried various settings and timings inside the controller function and restarted the system each time, no joy.

### Expected behavior

I'd like the outputs to run.

### Screenshots
[image]

### Additional context
Thanks again for this excellent piece of software; it's been a Godsend to me.

kizniche, 2022-10-07:
Thanks for the bug report. I'll take a look into it.

kizniche, 2022-10-07:
I couldn't replicate the issue.

```
2022-10-07 14:53:13,000 - INFO - mycodo.controllers.controller_pid_ecb6609d - Activated in 142.9 ms
2022-10-07 14:53:13,094 - DEBUG - mycodo.controllers.controller_pid_ecb6609d - Latest (CH0, Unit: pH): 6.055167607664593 @ 2022-10-07 14:53:06
```

As you can see from:

https://github.com/kizniche/Mycodo/blob/54ec9d8562609b35690d2929179a890b68615e6b/mycodo/controllers/controller_pid.py#L451-L454

My Daemon log shows the last measurement timestamp/value being properly parsed.

kizniche, 2022-10-07:
Are you using influxdb version 1.x or 2.x? 1.x is the recommended version since it has the most testing. v2.x has only been introduced as an option recently and may have some issues that need fixing.

sagehenstudio, 2022-10-07:
> Are you using influxdb version 1.x or 2.x?

OK, this is a helpful tip. I haven gone back and forth between the two in the "Configure" settings, but don't know if I need to change anything other than that. Also, I don't have access from localhost:8086 as suggested I might at:
 
[image]

and FWIW the db stuff on the Pi is over my head. I can poke around with SSH some, but that's about it. I'm a PHP/MySQL gal. Thank you.

sagehenstudio, 2022-10-07:
With 1.x, the error message becomes:

`2022-10-07 13:17:19,607 - ERROR - mycodo.controllers.controller_pid_9be1dfa4 - Exception while reading measurement from the influxdb database
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 439, in get_last_measurement_pid
    self.last_measurement = read_influxdb_single(
  File "/var/mycodo-root/mycodo/utils/influx.py", line 172, in read_influxdb_single
    data = query_string(
  File "/var/mycodo-root/mycodo/utils/influx.py", line 95, in query_string
    ret_value = query_string_influx_1(
  File "/var/mycodo-root/mycodo/utils/influxdb_1.py", line 240, in query_string_influx_1
    raw_data = dbcon.query(query).raw
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb/client.py", line 521, in query
    response = self.request(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb/client.py", line 378, in request
    raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 401: {"code":"unauthorized","message":"Unauthorized"}`

Unauthorized? Who me? Do you have any tips for me for what do do next? I suppose I could do a fresh install from scratch again, but seems like it doesn't hurt to ask in case you have a hunch for what to do without trying that again.

kizniche, 2022-10-07:
You can uninstall 2.x and install 1.x via apt on the command line and not have to reinstall Mycodo/OS. I'm currently on the road, so I can't provide any commands, but may get a chance in a few hours, if I have internet access where I'm going.

sagehenstudio, 2022-10-07:
That's very kind of you, but I've already gone ahead with a complete re-build and am almost done with that. Maybe post the command line here in case someone ends up in this same situation, if you get a chance. Fingers crossed, and note to self, bigger isn't always better ( 2.x < 1.x)

kizniche, 2022-10-31:
Reopning until the bug is fixed.

closed 2023-01-14 by kizniche