kizniche/Mycodo #1233: No module named 'RTIMU' error when using Sense Hat

wingsergey, 2022-10-08:
### Describe the problem/bug
Sense hat input is not working.

ERROR - mycodo.inputs.sense_hat_0906d318 - Error 101: Device not set up.

### Versions:

 - Mycodo Version: *8.14.2*
 - Raspberry Pi Version: *4B+*
 - Raspbian OS Version: *Bullseye Desktop*

### Additional context

2022-10-08 10:21:56,047 - ERROR - mycodo.inputs.sense_hat_0906d318 - Initialization errored 3 times; giving up. Maybe the following traceback can help diagnose the issue.
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/abstract_base_controller.py", line 52, in try_initialize
    self.initialize()
  File "/home/pi/Mycodo/mycodo/inputs/sense_hat.py", line 120, in initialize
    from sense_hat import SenseHat
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sense_hat/__init__.py", line 2, in <module>
    from .sense_hat import SenseHat, SenseHat as AstroPi
  File "/var/mycodo-root/env/lib/python3.9/site-packages/sense_hat/sense_hat.py", line 10, in <module>
    import RTIMU  # custom version
ModuleNotFoundError: No module named 'RTIMU'

I tried https://forums.raspberrypi.com/viewtopic.php?t=197281 and it not helped.

kizniche, 2022-10-10:
See if the Input works after this command:

`sudo ~/Mycodo/env/bin/pip install rtimulib`

If it does, I'll add that as a dependency for the Input.

wingsergey, 2022-10-11:
Unfortunately it failed:
```
pi@rpi:~/Mycodo` $ sudo ~/Mycodo/env/bin/pip install rtimulib
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rtimulib
  Using cached RTIMULib-7.2.1.tar.gz (7.2 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: rtimulib
  Building wheel for rtimulib (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      running bdist_wheel
      running build
      running build_ext
      building 'RTIMU' extension
      creating RTIMULib
      creating RTIMULib/IMUDrivers
      creating build
      creating build/temp.linux-aarch64-cpython-39
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTHumidity.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTHumidity.o -std=c++0x
      aarch64-linux-gnu-gcc: error: ../../RTIMULib/IMUDrivers/RTHumidity.cpp: No such file or directory
      aarch64-linux-gnu-gcc: fatal error: no input files
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for rtimulib
  Running setup.py clean for rtimulib
Failed to build rtimulib
Installing collected packages: rtimulib
  Running setup.py install for rtimulib ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for rtimulib did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      running install
      /home/pi/Mycodo/env/lib/python3.9/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_ext
      building 'RTIMU' extension
      creating build
      creating build/temp.linux-aarch64-cpython-39
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTHumidity.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTHumidity.o -std=c++0x
      aarch64-linux-gnu-gcc: error: ../../RTIMULib/IMUDrivers/RTHumidity.cpp: No such file or directory
      aarch64-linux-gnu-gcc: fatal error: no input files
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> rtimulib

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

kizniche, 2022-10-12:
Try these commands:

```bash
cd
git clone https://github.com/RPi-Distro/RTIMULib
cd ./RTIMULib/Linux/python/
~/Mycodo/env/bin/python setup.py build
~/Mycodo/env/bin/python setup.py install
```

hugocollignon, 2023-01-05:
Hello, and thanks for your work

I had the same issue, and when I tried to run by hand commands (from here https://github.com/kizniche/Mycodo/blob/master/mycodo/inputs/sense_hat.py#L98):

```
pi@raspberrypi:/tmp/RTIMULib/Linux/python $ ~/Mycodo/env/bin/python setup.py build
running build
running build_ext
building 'RTIMU' extension
creating RTIMULib
error: could not create 'RTIMULib': Permission denied
```

It's ok with sudo
```
pi@raspberrypi:/tmp/RTIMULib/Linux/python $ sudo ~/Mycodo/env/bin/python setup.py build
running build
running build_ext
building 'RTIMU' extension
creating RTIMULib
creating RTIMULib/IMUDrivers
creating build
creating build/temp.linux-aarch64-cpython-39
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTHumidity.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTHumidity.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTHumidityHTS221.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTHumidityHTS221.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTHumidityHTU21D.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTHumidityHTU21D.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTIMU.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTIMU.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTIMUBMX055.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTIMUBMX055.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTIMUBNO055.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTIMUBNO055.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTIMUGD20HM303D.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTIMUGD20HM303D.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTIMUGD20HM303DLHC.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTIMUGD20HM303DLHC.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTIMUGD20M303DLHC.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTIMUGD20M303DLHC.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/pi/Mycodo/env/include -I/usr/include/python3.9 -c ../../RTIMULib/IMUDrivers/RTIMULSM9DS0.cpp -o build/temp.linux-aarch64-cpython-39/../../RTIMULib/IMUDrivers/RTIMULSM9DS0.o -std=c++0x
aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-secu [... cut ...]
Installed /home/pi/Mycodo/env/lib/python3.9/site-packages/RTIMULib-7.2.1-py3.9-linux-aarch64.egg
Processing dependencies for RTIMULib==7.2.1
Finished processing dependencies for RTIMULib==7.2.1
```

And by the way, with the Sense HAT v2, we have a color sensor included now!
See: https://datasheets.raspberrypi.com/sense-hat/sense-hat-product-brief.pdf
And https://github.com/astro-pi/python-sense-hat/pull/131 to patch the library (the light sensor on modules shipped in ISS are not the same than those selled).

Thanks!

closed 2023-01-14 by kizniche