kizniche/Mycodo #1234: Sensiron SCD40 Force CO2 Calibration Failing.

gngenius02, 2022-10-16:
### Describe the problem/bug

When trying to force CO2 calibration on Sensiron SCD40. It shows as command sent, but it kills the sensor and I have to Deactivate and Activate it again to get it to start working. I find the following errors repeating in the logs about every 20 seconds. 

```
2022-10-16 00:59:25,059 - ERROR - mycodo.inputs.scd4x_circuitpython_a5664e17 - Error setting CO2 Concentration: unsupported operand type(s) for >>: 'float' and 'int'
2022-10-16 00:59:35,198 - ERROR - mycodo.inputs.scd4x_circuitpython_a5664e17 - data_ready is False
2022-10-16 00:59:45,336 - ERROR - mycodo.inputs.scd4x_circuitpython_a5664e17 - data_ready is False
2022-10-16 00:59:55,487 - ERROR - mycodo.inputs.scd4x_circuitpython_a5664e17 - data_ready is False
2022-10-16 00:59:55,487 - ERROR - mycodo.controllers.controller_input_a5664e17 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
```

### Versions:

 - Mycodo Version: 8.14.2
 - Raspberry Pi Version: 4
 - Raspbian OS Version: Raspbian GNU/Linux 11 (bullseye)

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to Inputs, then click the Configuration button for SCD-4x
2. On modal, there is a Calibrate CO2 button.
3. I type in the CO2 value I want to calibrate. I tried 400(int) and 400.0(float) as entries. 
4. I press the "Calibrate CO2" button
5. It gives me a "Success: Custom Button:  Command sent to Input Controller and is running in the background."
6. The device never gets readings again and logs show errors as stated above and in screenshot shown below.

### Recovery From Issue.

In order to get readings from the sensor again:

1. I need to go into inputs.
2. For that device I need to click on Deactivate.
3. Wait till Deactivated.
4. Finally click Activate.

### Expected behavior

I believe it should calibrate the sensor to the value in the input box. Then after calibration complete it should start displaying that value as the current value.

### Screenshots

Here is the Input location I have circled the location of the calibration area.
[image]

This is the errors that show up in the logs.
[image]

### Additional context

I am guessing the issue is between different brand of sensors, maybe coming from the adafruit library? I couldn't find anything I could fix in the MyCodo code. I know the manufacturers shouldn't make a difference but maybe one of them is handling types differently. If you need more information or data or to check anything specific please let me know. I will dive into the adafruit lib now to see if I can figure out where the issue is.

gngenius02, 2022-10-16:
I actually just verified the library is expecting an `int` as can be seen here:
https://github.com/adafruit/Adafruit_CircuitPython_SCD4X/blob/a9ae07d8c30c45e7a36d5308bdaaa2887c3c7e31/adafruit_scd4x.py#L161

The code is passing a float. Should be an easy fix.

gngenius02, 2022-10-16:
Making that change locally did not fix the issue. Not sure if I have to restart or maybe I have the install located somewhere else or not. Will investigate further at a future time.

kizniche, 2022-10-16:
Nice work. Thanks for figuring out the issue and making a PR.

closed 2022-12-24 by gngenius02

kizniche, 2023-01-14:
This pull request has been mentioned on **Radical DIY Forum**. There might be relevant details there:

https://forum.radicaldiy.com/t/mycodo-v8-15-0-release/1362/1