kizniche/Mycodo #1237: Error when recovering from 8.13.09 to 8.14.2

VitorFrost, 2022-10-25:
Also change from raspberry 3B+ to 4B

STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

*[A clear and concise description of what you were trying to do and what the problem/bug is - replace this with your own text]*

### Versions:

 - Mycodo Version: *[e.g. 8.0.3]*
 - Raspberry Pi Version: *[e.g. 3B+]*
 - Raspbian OS Version: *[e.g. Buster Lite]*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *[Go to...]*
2. *[Click on...]*
3. *[Scroll down to...]*
4. *[See error...]*
5. *[replace this list with your own text]*

### Expected behavior

*[A clear and concise description of what you expected to happen - replace this with your own text]*

### Screenshots

*[If applicable, add screenshots to help explain your problem - replace this with your own text]*

### Additional context

*[Anything that should be added to make it easier to address this issue - replace this with your own text]*

kizniche, 2022-10-26:
Are there any additional relevant log lines in /var/log/mycodo/mycodorestore.log?

kizniche, 2022-10-26:
There shouldn't be any issue restoring a system that has a higher version number than the one installed, since the database is supposed to be upgraded in the restore process. But it appears your database was not upgraded properly. You can attempt to run the post_upgrade script again, which includes the database upgrade command:

`sudo ~/Mycodo/mycodo/scripts/upgrade_post.sh`

Even if that works, I would still like to see what happened in your restore log. Alternatively, it's much safer to install the same version of Mycodo as the export file, then import that settings file.

closed 2022-11-22 by kizniche