kizniche/Mycodo #1242: Proper way to multiplex a set of sensors (multiple reservoirs) to target multiple influxDB buckets

nargetdev, 2022-11-23:
We've built a solenoid manifold which routes water to three separate reservoirs.  Effectively you use one set of probes to sample as many reservoirs as you like, and one set of peristaltic to dose all reservoirs and maintain distinct EC, pH levels for each.  Of course with the caveat that some logic is needed to route that one set of sensor's info to three separate influxDB buckets.  Furthermore, special care must be taken so that any closed loop control algorithm must only run when the pump is actively sampling.

**Describe alternatives you've considered**
It is actually possible to implement a configuration like this in existing Mycodo, but it's not pretty. For example, three separate pH inputs (all wired to the same physical sensor), and having separate "daily time span/point" functions which turn on and off those inputs to pull the sample and separately both activate the correct control loop algo at the correct time - one for each reservoir to maintain separate set-points.

**Additional context**
I'm considering implementing this entirely externally by configuring Telegraf to switch buckets on the backend from an MQTT message, and then use a golang process to execute control loops by reading the influxdb stream and leverage the Mycodo API to actually execute the outputs dosing.  This feels cleaner to me.

Just curious if anybody has thought about this problem.

kizniche, 2022-11-24:
I just refactored the Influxdb 1.x code to use Flux (master branch), so now the influxdb code is consolidated and very consistent with using Flux to access both 1.x and 2.x databases. This should help to move toward being able to specify different buckets. How that materializes, I'm not sure, so I'll have to chew on it for a while.

nargetdev, 2023-01-20:
That was actually how I did it initially.  But my intention was to soon after go to 8 reservoirs, and it quickly got unwieldy, as I also had functions which needed to shut off the outputs to many GPIO to configure the sample loop.

closed 2023-01-20 by nargetdev