kizniche/Mycodo #1247: Support for SparkFun Qwiic Relays?

victorhooi, 2022-12-05:
**Is your feature request related to a problem? Please describe.**

Currently, it seems MyCodo only supports a single type of relay - the Seeed Grove relays:

https://wiki.seeedstudio.com/Grove-Relay/

**Describe the solution you'd like**

It would be nice if MyCodo could support other types of commonly available relays - e.g. the Sparkfun Qwiic relays:

* [https://www.sparkfun.com/products/16810](https://www.sparkfun.com/products/16810)
* [SparkFun Qwiic Quad Solid State Relay Kit](https://www.sparkfun.com/products/16833)
* [SparkFun Qwiic Single Relay](https://www.sparkfun.com/products/15093)
* [SparkFun Qwiic Quad Relay](https://www.sparkfun.com/products/16566)

(I'm intending to get some of the quad solid state relay kits).

These normally have pretty good availability, and are well supported by SparkFun, with good documentation - e.g.:

https://learn.sparkfun.com/tutorials/sparkfun-qwiic-quad-solid-state-relay-kit-hookup-guide/qwiic-relay-python-package

They also provide easy connectivity with Qwiic (basically JST-SH).

Here is the Github repo for the Raspberry Pi python libraries for the relays:

https://github.com/sparkfun/Qwiic_Relay_Py

**Describe alternatives you've considered**
Use the existing Seeed Grove relays - however, the kits for these seem a bit dinky (although they do have an include case, even if all the screws/standoffs are 3D printed PLA).

**Additional context**
Add any other context or screenshots about the feature request here.

kizniche, 2022-12-05:
There doesn't need to be any special code to support the use of a relay beyond merely switching a GPIO pin high/low. In this regard, Mycodo supports thousands of relays. What you're referring to is support for relays that have additional control circuitry, such as being able to be controlled via I2C or other communication protocols. Therefore, using just the GPIO Output you can control any relay that can be switched with 3.3 VDC as long as the switching current doesn't exceed the capacity of the GPIO pin or cumulative capacity of all GPIOs.

That being said, I do think Sparkfun makes quality products and the ones you linked to look like great options for implementing relays without having to design your own circuits. I don't have any of the products you listed, but I'd be happy to help develop new Output modules for them, if you or anyone else that has them would be willing to test.

victorhooi, 2022-12-13:
In this case, the Sparkfun Qwiic relays are controlled via I2C (and I believe you can daisy-chain multiple ones).

I've ordered a couple to test with - once I get them, what could I do to help figure out how to support them?

victorhooi, 2022-12-17:
My SparkFun order with these relays arrived:

[image]

Let me know if there's anything I can do to help test, or gather any info you might need etc.

kizniche, 2022-12-17:
I just committed a new Output module for the 4-relay Sparkfun boards. You can test the latest code by [upgrading to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch). If it works, I'll add additional Output modules for the 2 relay and 1 relay boards.

https://github.com/kizniche/Mycodo/commit/c120e42159fdc952ad8be210a4eddc3b937d47d7