kizniche/Mycodo #1249: pylint is taking way too much time in Analyzing the code

summa-code, 2022-12-07:
On edge devices with low processing powers, pylint is taking way too much time. Is there a way to configure this such that I can enable/disable warnings and others ? I think that Errors are the most important thing for edge devices.

Not sure if I can contribute, but is there a specific place in the code that I can try and play around a little bit ?

kizniche, 2022-12-07:
I agree, a status message indicating it's processing is also a good idea. I'll see if I can implement it soon.

kizniche, 2022-12-20:
I just added the option to Analyze with Pylint to modules that use Python code (Inputs, Outputs, and Conditional Functions) in https://github.com/kizniche/Mycodo/commit/705664fe27eeb194c97d37d014da363895a1247b.

closed 2023-01-14 by kizniche