kizniche/Mycodo #1250: Cannot export from DB - unauthorised

cperricos, 2022-12-11:
STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues

Please DO NOT OPEN AN ISSUE:

 - If your Mycodo version is not the latest release version, please update your device before submitting your issue (unless your issue is related to not being able to upgrade). Your problem might already be solved.

 - If your issue has been addressed before. If you have any new information that may aid in solving the issue, post it in the issue that already exists.

If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

Hello, Awesome software!
I have had  a number of small issues with my installation - live measurements wont show (despite being able to publish measurements on mqtt). i think there is a problem with getting authorisation to read/write from the DB. When I try and export a particular measurement from the DB, 

### Versions:

MyCodo Version: 8.14.2
Database: 07c7c8ebc195
Model: Raspberry Pi 3 Model B Rev 1.2
Release:
Distributor ID: Debian
Description: Debian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Goto More->Export/Import
2. Click on Export Data As CSV
3. 3. Select measurement (I selected Temp RTD as I can see the sensor is working and publishing value on MQTT)

### Expected behavior

This is the screen text I get:

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a [new issue on GitHub](https://github.com/kizniche/Mycodo/issues/new?template=problem-bug-report.md) with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.14.2
Database: 07c7c8ebc195
Model: Raspberry Pi 3 Model B Rev 1.2

Release:
Distributor ID: Debian
Description: Debian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

Firmware:
b''

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_general.py", line 369, in export_data
    data = query_string(
  File "/home/pi/Mycodo/mycodo/utils/influx.py", line 95, in query_string
    ret_value = query_string_influx_1(
  File "/home/pi/Mycodo/mycodo/utils/influxdb_1.py", line 240, in query_string_influx_1
    raw_data = dbcon.query(query).raw
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb/client.py", line 521, in query
    response = self.request(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb/client.py", line 378, in request
    raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 401: {"code":"unauthorized","message":"Unauthorized"}

kizniche, 2022-12-21:
Are you using influxdb 1.x or 2.x?

cperricos, 2022-12-21:
Hi, thanks for the reply. I was using 2.x but switched to 1.x but it's all working. Will now try and write the whole code around managing water quality in Mycodo so fingers crossed!

kizniche, 2023-01-04:
Duplicate of #1232

closed 2023-01-04 by kizniche