kizniche/Mycodo #1252: Honeywell HIH6130 temperature/humisity sensor

atlas55x, 2022-12-18:
Hi Kixzniche,

I have a honeywell HIH6130-based sensor from Sparkfun for both temp/relative humidity. Is there a similar i2C input that will work? I would love to be able to get it to work, but it is outside of my capabilities. If it's not something that is easy, I will purchase an already-supported sensor. I am 3 days into Mycodo and it is so cool!

Thanks!
-Atlas55x

https://www.amazon.com/Humidity-Temperature-Sensor-Breakout-HIH6130/dp/B00L8EB7YK

kizniche, 2022-12-18:
This doesn't seem to be a product that Sparkfun still supports. I'd recommend finding something that they're still selling. There are many temperature/humidity sensors supported by Mycodo that are currently still for sale by the manufacturer or breakout board-makers.

atlas55x, 2022-12-18:
Thanks for the fast response! Adafruit Sensiron SHT31-D is on the way. Happy Holidays.

closed 2022-12-18 by atlas55x