kizniche/Mycodo #1254: Cannot get Ezo pump to work with bang bang

cperricos, 2022-12-30:
### Describe the problem/bug

So I am making sme progress in my pool controller, but cant get my EZO pump to work with the bang bang controller. I set a threshold of 750mv ORP which should activate the pump (it wokrs when I try to dispense etc via the out put controller)

### Versions:

 - Mycodo Version: 8.14.2
 - Raspberry Pi Version: 3B
 - Raspbian OS Version: *[e.g. Buster Lite]*

the log gives the following errors:

[Action 0da1bcb1, MQTT: Publish]: MQTT Publish '220'.
2022-12-30 13:41:30,039 - ERROR - mycodo.inputs.test_00_cbc058f4 - Exception: 'NoneType' object has no attribute 'randint'
2022-12-30 13:41:30,040 - ERROR - mycodo.controllers.controller_input_cbc058f4 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-12-30 13:41:35,982 - DEBUG - mycodo.function.bang_bang_8892780e - Input: 191.45000000000002, output: off, target: 750.0, hyst: 30.0
2022-12-30 13:41:35,989 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_642e7235 - output_on_off(on, 0, None, 0.0, 0.0, True)
2022-12-30 13:41:36,017 - ERROR - mycodo.outputs.pump_atlas_ezo_pmp_642e7235 - Invalid parameters: State: on, Type: sec, Mode: specify_flow_rate, Amount: None, Flow Rate: 50.0
2022-12-30 13:41:36,018 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_642e7235 - Output 642e7235-99ee-47ec-9cb3-5636a98b2291 CH0 (Acid) ON at 2022-12-30 13:37:05.985332. Output returned: None
2022-12-30 13:41:44,994 - ERROR - mycodo.inputs.test_00_cbc058f4 - Exception: 'NoneType' object has no attribute 'randint'
2022-12-30 13:42:00,037 - ERROR - mycodo.inputs.test_00_cbc058f4 - Exception: 'NoneType' object has no attribute 'randint'
2022-12-30 13:42:05,902 - DEBUG - mycodo.function.bang_bang_8892780e - Input: 191.45000000000002, output: off, target: 750.0, hyst: 30.0
2022-12-30 13:42:05,910 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_642e7235 - output_on_off(on, 0, None, 0.0, 0.0, True)
2022-12-30 13:42:05,933 - ERROR - mycodo.outputs.pump_atlas_ezo_pmp_642e7235 - Invalid parameters: State: on, Type: sec, Mode: specify_flow_rate, Amount: None, Flow Rate: 50.0
2022-12-30 13:42:05,933 - DEBUG - mycodo.outputs.pump_atlas_ezo_pmp_642e7235 - Output 642e7235-99ee-47ec-9cb3-5636a98b2291 CH0 (Acid) ON at 2022-12-30 13:37:05.985332. Output returned: None
2022-12-30 13:42:14,996 - ERROR - mycodo.inputs.test_00_cbc058f4 - Exception: 'NoneType' object has no attribute 'randint'
2022-12-30 13:42:14,997 - ERROR - mycodo.controllers.controller_input_cbc058f4 - StopIteration raised 3 times. Possibly could not read input. Ensure it's connected properly and detected.
2022-12-30 13:42:28,241 - INFO - mycodo.controllers.controller_conditional_934faf83 - This INFO log entry will appear in the Daemon Log
2022-12-30 13:42:28,297 - INFO - mycodo.controllers.controller_conditional_934faf83 - Measurement value is 1
2022-12-30 13:42:28,529 - DEBUG - mycodo.action.mqtt_publish_0da1bcb1 - Message: 2022-12-30 13:42:28

kizniche, 2023-01-11:
I just pushed a potential fix. If you would like to [upgrade to master](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and test, please let me know if it fixes the issue.

kizniche, 2023-01-14:
Mycodo 8.15.0 has been released, and incorporates this untested fix. Please let me know if the issue persists after upgrading.

closed 2023-01-20 by kizniche