kizniche/Mycodo #1255: exceptions thrown in regulate_ph_ec.py

cperricos, 2023-01-02:
Hello,
I am trying to use the "Regulate PH and EC" function from your library to regulate PH and ORb in my pool system (I think the function is amaing and just what i need). I can get it to work with my Ezo pump however. I have tried parameter that will actvate the pmp, but the code causes excpetions here is the log:

2023-01-02 09:28:34,517 - INFO - mycodo.controllers.controller_function_dfc57aca - Activated in 1723.7 ms
2023-01-02 09:28:44,198 - DEBUG - mycodo.function.regulate_ph_ec_dfc57aca - Most recent timestamp and measurement for pH: 2023-01-02T09:28:36.426246Z, 7.074249999999999
2023-01-02 09:28:44,575 - DEBUG - mycodo.function.regulate_ph_ec_dfc57aca - Most recent timestamp and measurement for EC: 2023-01-02T09:28:26.837841Z, 840.3833333333332
2023-01-02 09:28:44,583 - DEBUG - mycodo.function.regulate_ph_ec_dfc57aca - pH is dangerously high: 7.07. Should be < 7.00. Dispensing 2.0 sec acid
2023-01-02 09:28:44,590 - ERROR - mycodo.controllers.controller_function_dfc57aca - Exception while running loop()
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_function.py", line 76, in loop
    self.run_function.loop()
  File "/home/pi/Mycodo/mycodo/functions/custom_functions/regulate_ph_ec.py", line 806, in loop
    "sec_ph_lower", self.get_custom_option("sec_ph_lower") + self.output_ph_amount)
TypeError: unsupported operand type(s) for +: 'NoneType' and 'float'

any suggestions?

many thanks

cperricos, 2023-01-11:
> You are using a custom imported Function rather than the   latest module on master (https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fkizniche%2FMycodo%2Fblob%2F7009c4237767dcc158620a4f81bad9cea5a4491b%2Fmycodo%2Ffunctions%2Fregulate_ph_ec.py&data=05%7C01%7C%7C889ad973c3e241754de708daf380a2c4%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638090031051862397%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=hEdPwrRIIqC%2Fxzdi15uJ1DoDXRK6lWQcNPcUhQew31g%3D&reserved=0).
>
>
>
> You should be using the latest code on master before submitting any but reports.

Apologies - will get the latest code

kizniche, 2023-01-13:
No problem. If you still have issues after you try the code on github, let us know here what errors you get.

cperricos, 2023-01-13:
I think I have managed to get it to work now.
  
Am exception is thrown when the timers and summation volumes aren’t reset at the beginning of the function. From what I can tell but otherwise it’s great.
  
One additional question I had is how can the timer and summation variables be accessed from outside the function?
  

  
For example, if we are summing    up the total amount of acid (Ph-) dispensed and we know the overall acid container volume you could keep track of how much acid is left in the container and send an alert when a new container needs to be ordered. Is it    possible To expose the summation measurements so they can be used in other functions or shown on a dashboard?
  

  
Thanks

kizniche, 2023-01-14:
>exception is thrown

What is the error?