kizniche/Mycodo #1256: Pi Camera issue

gab16, 2023-01-02:
threw an error when trying to add an enabled Pi camera connected to Pi4B via ribbon cable

Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a [new issue on GitHub](https://github.com/kizniche/Mycodo/issues/new?template=problem-bug-report.md) with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.14.2
Database: 07c7c8ebc195
Model: Raspberry Pi 4 Model B Rev 1.4

Release:
Distributor ID: Debian
Description: Debian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

Firmware:
b''

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 271, in page_camera
    camera_stream = import_module(
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/mycodo_flask/camera/camera_picamera.py", line 7, in <module>
    import picamera
  File "/var/mycodo-root/env/lib/python3.9/site-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/var/mycodo-root/env/lib/python3.9/site-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/var/mycodo-root/env/lib/python3.9/site-packages/picamera/mmal.py", line 49, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory

kizniche, 2023-01-15:
Camera streaming has always been very buggy. It looks like you're running older code, but nonetheless I see a try/except block should be added to catch this error rather than crashing the page as it does. I've added this error-handling to the latest commit on the master branch, in commit https://github.com/kizniche/Mycodo/commit/e4fd40b8b1f10daafc529442eae682e32aa9f638.

gab16, 2023-03-03:
Sounds good will try the mod, thanks!