kizniche/Mycodo #1258: Issue with displaying Input Readings

berserktron3k, 2023-01-06:
Clean install of Mycodo 8.14.2 on a Raspberry Pi 3B+. I'm able to add 2 inputs (SHT3x), however when going to Live Measurements (or any other location to display measurements) it only returns zeros.

Added two of these ([FS200-SHT3X](https://www.amazon.com/dp/B07Z4JGDQK?psc=1&ref=ppx_yo2ov_dt_b_product_details)) and had to create a second I2C bus since they both use 0x44 as their address. Added this to /boot/config.txt to enable another I2C bus: `dtoverlay=i2c-gpio,bus=3,i2c_gpio_delay_us=1,i2c_gpio_sda=23,i2c_gpio_scl=24` 

Enabled debug on the two inputs and can see that it seems to be accurately pulling data, as shown here from /var/log/mycodo/mycodo.log:
`2023-01-06 16:20:46,282 - DEBUG - mycodo.controllers.controller_input_644cbdf5 - Adding measurements to InfluxDB with ID 644cbdf5-77d0-4870-b1a1-243bfe537381: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 18.18532082093538, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 20, 46, 256284)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 43.79797055008774, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 20, 46, 256309)}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 5.6570107609912, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 20, 46, 256382)}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1173.346463778754, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 20, 46, 256430)}}
2023-01-06 16:20:46,498 - DEBUG - mycodo.controllers.controller_input_f5271d37 - Adding measurements to InfluxDB with ID f5271d37-4a74-4346-8cc4-b169f63cdd55: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 19.15465018692302, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 20, 46, 455808)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 41.104753185320824, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 20, 46, 455830)}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 5.615484302443015, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 20, 46, 455894)}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1306.4256466873112, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 20, 46, 455941)}}
2023-01-06 16:21:01,265 - DEBUG - mycodo.controllers.controller_input_644cbdf5 - Adding measurements to InfluxDB with ID 644cbdf5-77d0-4870-b1a1-243bfe537381: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 18.18532082093538, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 21, 1, 210198)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 43.78728923476005, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 21, 1, 210227)}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 5.6534869593041925, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 21, 1, 210294)}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1173.5694607717614, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 21, 1, 210347)}}
2023-01-06 16:21:01,502 - DEBUG - mycodo.controllers.controller_input_f5271d37 - Adding measurements to InfluxDB with ID f5271d37-4a74-4346-8cc4-b169f63cdd55: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 19.168001831082634, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 21, 1, 474003)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 41.12001220721752, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 21, 1, 474027)}, 2: {'measurement': 'dewpoint', 'unit': 'C', 'value': 5.632854119651711, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 21, 1, 474090)}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1307.1741404754723, 'timestamp_utc': datetime.datetime(2023, 1, 6, 22, 21, 1, 474138)}}"`

However when I go to Live Measurements, here is what I get:
[image]

Please let me know any additional information you may need, and as always, thanks for the help!

kizniche, 2023-01-13:
There's currently a bug in the latest release when using influxdb 2. It's fixed in the yet-to-be released version, I just need to finalize and release it.

kizniche, 2023-01-14:
Mycodo 8.15.0 was just released and fixes the issue with influxdb.

closed 2023-01-14 by kizniche