kizniche/Mycodo #1264: Installation on ver. 8.15.1 crashes.

Mithred, 2023-01-18:
New to Mycodo, managed to get everything working on 8.14.2 last week. Accidentally upgraded Mycodo to 8.15.1. I formatted and reinstalled with 8.15.1, issue persists. 

On one of the last steps, mycodo installation crashes at "restarting mycodo daemon". You can login to the web interface at this point, however daemon is not running. 

mycodo.log is empty. 

From my knowledge, I believe it is due to a configuration error on InfluxDB. 

### Versions:

 - Mycodo Version: *8.15.1*
 - Raspberry Pi Version: *Zero W*
 - Raspbian OS Version: *Buster Lite / Bullseye Lite*

kizniche, 2023-01-18:
Can you open a terminal, run the following command, and copy the output here?

`sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py`

kauaiquinlan, 2023-01-18:
Im having the same issue, Thanks for the help and response.

kauaiquinlan, 2023-01-18:
Putty would not let me execute the full command.

Mithred, 2023-01-18:
I have manually installed 8.14.2 which is working perfectly as expected. 

I have installed 8.15.1 onto a different SD Card, using your command in a terminal, the output was (after a minute or two): 

"Unable to lock on the pidfile."

Thank you for your help, the code and support Kyle :)

kizniche, 2023-01-18:
The daemon PID file is located at /var/run/mycodo.pid and since this is tempfs, it is deleted at reboot. So there can't be any persistent issues with locked/stale PID files. Did you attempt a reboot to alleviate the PID issue?

kauaiquinlan, 2023-01-18:
Hello Team, I was wondering where to locate version 8.14.2 to install instead of the latest version? 
I appreciate the help 🙏🏻🙌🏻

kizniche, 2023-01-18:
>Putty would not let me execute the full command.

What do you mean by this? Putty only makes the connection, it has no ability to affect commands.

kauaiquinlan, 2023-01-18:
Hey Kyle, Thanks for the response. Let me clarify. Above when you said, open a terminal and run the command, 

sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py

I was not getting an output when I entered it into PuTTy. When I entered the command, 

sudo ~/Mycodo/env/bin/python  

I got an output which I sent to you in the above comment. My wording was not correct which I apologize. I am fairly green in this programming space but appreciate the dialogue. I was wondering what would be my next step to get a running mycodo system? Should I re-install mycodo from github? Im not sure how to follow your directions on locating the daemon PID file to resolve the issue. I see mithred installed a previous version of mycodo, but Im not sure how to manually download a previous version. I appreciate your time and help! Id love to send you pictures of my system when its up and running. You are a big part of its success. Cheers brother and have a great day!

kizniche, 2023-01-19:
I wouldn't recommend installing an old version, as the latest release should install and run. There are dozens if not hundreds of systems currently on the 8.15.x release and running fine, so what you're experiencing is a fluke of some kind. What influxdb version did you select when you went through the installer? I would recommend starting fresh with a fresh install of Raspberry Pi OS and install Mycodo one more time and see if the issue is still there. If it occurs again, lets then start investigating further.

kizniche, 2023-01-19:
Also, @kauaiquinlan I see you're not the OP... Can you provide more information about your system (what board, what operating system)?

turgidpotato, 2023-01-19:
Hi, I have the same version and pi as the person who posted this issue.  Can you explain why the influxdb version may be the problem?  I tried to reinstall like you said but it downloaded to version 8.15.1 should I simply just start from scratch again with this new 15.3, and if the influxdb will continue to cause this problem if I try again?  Thank you so much again this program has taught me so much and I cannot thank you for your continued work on this project.

kizniche, 2023-01-19:
I never said influxdb would cause any problem, I merely asked which version was selected to be installed. You also haven't even said you experienced a problem or what that problem is. This issue should be for one specific issue, related to the original post, not a catchall area for anyone wishing to get support.

turgidpotato, 2023-01-19:
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 675, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 290, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/mycopitt/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

turgidpotato, 2023-01-19:
I tried to update from the exact same version that the person with this problem had.  
Sorry I don't know if the traceback would help.  This issue is exactly what I am experiencing just trying to help/ learn sorry not that well versed in this as well.  Thanks again...

kizniche, 2023-01-19:
No, your error is not the same as the original poster. See https://github.com/kizniche/Mycodo/issues/1265

kizniche, 2023-01-19:
I'm closing this issue, as it's become too disorganized. @Mithred and @kauaiquinlan, if you still have your issues, please reinstall your operating system and Mycodo, and if your issue still persists, open a new issue with as much detail about it as possible.

closed 2023-01-19 by kizniche