kizniche/Mycodo #1265: Fix for "AttributeError: 'NoneType' object has no attribute 'split'"

kizniche, 2023-01-19:
If you experience this error on the upgrade page or in your daemon log, you can fix it with the following command:

`sed -i 's/v\.\*(\\d\\\.\\d\\\.\\d)/v(\\d+)\\\.(\\d+)\\\.\\d+/g' ~/Mycodo/mycodo/utils/github_release_info.py && sudo service mycodoflask restart`

kizniche, 2023-01-19:
@turgidpotato, sorry for being curt with you earlier, it's been stressful dealing with this issue that has affected many users.

If this issue is affecting you and the above solution does not work, please comment and we will work on a solution. If it does work, please do not comment, as it will only fill this issue with clutter and distract from the issue.

closed 2023-01-19 by kizniche

turgidpotato, 2023-01-19:
No I totally get it the amount of stress this has caused me is probably a drop in a bucket compared to what you have been dealing with.  I have been really meaning to sincerely tell you thank you so much for creating this though, this has taught me so much and this is the first time I have really had any issues.  It is incredible how responsive you are and it really means the world that you responded so fast.  I love this program and I am light years behind you in coding prowess so I apologize.  But this project along with my love for mushroom cultivation has inspired me to continue into lab research in neuroscience and I one day hope to use the vast amount of polysacharide and phenolic compounds to find treatments for neurodegenerative diseases like Parkinsons.  Without you and this community I would not be where I am at today sincerely it has relit my passion for science when I was at a really low place.

kizniche, 2023-01-19:
That's good to hear and I'm glad it's helped you in some way. I wish you well in your endeavors and if you ever want to discuss non-issues related to Mycodo or automation, etc., come by the forum at https://forum.radicaldiy.com