kizniche/Mycodo #1267: Asynchronous Graphs results in a empty graph if the unit are kWh.

putte110, 2023-01-20:
Asynchronous Graphs from  kWh results in a empty graph.

 - From Mycodo Version: 8.15.1
 - Raspberry Pi Version: 3B+
 - Raspbian OS Version: Raspberry pi os(64-bit)

I had version 8.14.2 running, to my great satisfaction, one of the inputs measures the electricity consumption (kWh).
I've found that since the update to version 8.15.1 Asynchronous Graphs from  the electricity consumption results in a empty graph.
I've changed the units of this input ,  from kWh to Angle(degrees), and the historical measurements reappear.
With the update to version 8.15.4, i had adjusted the units back from(Angle to kWh), and it didn't work again.

kizniche, 2023-01-21:
Please provide detailed steps to replicate the behavior you're experiencing.

putte110, 2023-01-22:
Thanks for your email

The program was running fine until I ran the update from Mycodo 8.14.2 to
Mycodo 8.15.1
I then ran with rpi 32 bit keyboard settings Belgium Mycodo 8.14.2.
After the update to Mycodo 8.15.1 my rpi has crashed.
I then reinstalled the system 32 bit and Import Mycodo Settings Data from
ZIP(Mycodo 8.14.2).
Everything worked well except the kWh hystories.
I let the system continue to run until a new update became available.

I then reinstalled the system 64 bit db2.x. and Mycodo version 8.15.4 and
Import Mycodo Settings Data from ZIP(Mycodo 8.15.1).
Everything worked well again except the kWh hystories.

Because the hystorics of watts and degrees of other inputs worked, I then
adjusted the unit from kWh to Angle(degrees) in the input settings of the
kWh meter, and the hystorics of kWh worked again, but then of course
indicated Angle(degrees).

If you need any more info I want to make a new test project next week and
send it to you, unfortunately I can't find Mycodo version 8.14.2 anymore.

Sincerely

putte110, 2023-01-30:
I reinstalled everything again and instead of reloading the backup I
programmed everything back.
Everything works as it should.
Apparently something went wrong while making the backup from Mycodo 8.14.2.
"Mycodo Settings Data from ZIP".

Sincerely