kizniche/Mycodo #1277: 500 Error on Data CSV Export

K1rdro, 2023-02-13:
### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. Go to More
2. Click on Export Import 
3. Click on Export Data as CSV
4. See error:

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a [new issue on GitHub](https://github.com/kizniche/Mycodo/issues/new?template=problem-bug-report.md) with the following diagnostic information and error traceback (copy the entire traceback):

Version: 8.15.6
Database: 16b28ef31b5b
Model: Raspberry Pi 3 Model B Rev 1.2

Release:
Distributor ID: Debian
Description: Debian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

Firmware:
b''

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 674, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 290, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/kirdro/Mycodo/mycodo/mycodo_flask/routes_general.py", line 387, in export_data
    data = query_string(
  File "/home/kirdro/Mycodo/mycodo/utils/influx.py", line 291, in query_string
    ret_value = query_flux(
  File "/home/kirdro/Mycodo/mycodo/utils/influx.py", line 276, in query_flux
    tables = client.query_api().query(query)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/client/query_api.py", line 203, in query
    response = self._query_api.post_query(org=org, query=self._create_query(query, self.default_dialect, params),
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/service/query_service.py", line 279, in post_query
    (data) = self.post_query_with_http_info(**kwargs)  # noqa: E501
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/service/query_service.py", line 305, in post_query_with_http_info
    return self.api_client.call_api(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 343, in call_api
    return self.__call_api(resource_path, method,
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 121, in __call_api
    self._signin(resource_path=resource_path)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 657, in _signin
    http_info = SigninService(self).post_signin_with_http_info()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/service/signin_service.py", line 74, in post_signin_with_http_info
    return self.api_client.call_api(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 343, in call_api
    return self.__call_api(resource_path, method,
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 173, in __call_api
    response_data = self.request(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 388, in request
    return self.rest_client.POST(url,
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/rest.py", line 311, in POST
    return self.request("POST", url,
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/rest.py", line 261, in request
    raise ApiException(http_resp=r)
influxdb_client.rest.ApiException: (401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.6.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Mon, 13 Feb 2023 19:37:22 GMT', 'Content-Length': '62'})
HTTP response body: {
	"code": "unauthorized",
	"message": "unauthorized access"
}

Separate, but possibly related: I'm also unable to see live data on the Live page. I can see from the logs that it is adding measurements (which is why I was attempting to see what was actually being written). 

Daemon log:
2023-02-13 13:47:49,016 - DEBUG - mycodo.controllers.controller_input_03549026 - Adding measurements to InfluxDB with ID 03549026-0408-47d6-bfe7-5f9822e64d0d: {0: {'measurement': None, 'unit': 'F', 'value': 146.0768, 'timestamp_utc': datetime.datetime(2023, 2, 13, 19, 47, 48, 922136)}, 1: {'measurement': None, 'unit': 'F', 'value': 146.12, 'timestamp_utc': datetime.datetime(2023, 2, 13, 19, 47, 48, 950025)}}
2023-02-13 13:47:57,080 - DEBUG - mycodo.controllers.controller_input_f7612bf9 - Adding measurements to InfluxDB with ID f7612bf9-c713-453b-ba21-e8defa940b69: {0: {'measurement': None, 'unit': 'F', 'value': 70.46304, 'timestamp_utc': datetime.datetime(2023, 2, 13, 19, 47, 55, 11609)}, 1: {'measurement': 'humidity', 'unit': 'percent', 'value': 37.465476462958726, 'timestamp_utc': datetime.datetime(2023, 2, 13, 19, 47, 55, 11634)}, 2: {'measurement': None, 'unit': 'F', 'value': 43.25468, 'timestamp_utc': datetime.datetime(2023, 2, 13, 19, 47, 55, 11700)}, 3: {'measurement': 'vapor_pressure_deficit', 'unit': 'Pa', 'value': 1590.4345854047826, 'timestamp_utc': datetime.datetime(2023, 2, 13, 19, 47, 55, 11757)}}
2023-02-13 13:48:01,128 - DEBUG - mycodo.controllers.controller_input_5eb8745d - Adding measurements to InfluxDB with ID 5eb8745d-5b37-4cff-8bbd-b49f99ba4ddf: {0: {'measurement': None, 'unit': 'F', 'value': 62.6, 'timestamp_utc': datetime.datetime(2023, 2, 13, 19, 48, 1, 99537)}}
2023-02-13 13:48:03,919 - DEBUG - mycodo.inputs.rpi_cpu_gpu_temperature_03549026 - CPU Temperature: 63.914
2023-02-13 13:48:03,948 - DEBUG - mycodo.inputs.rpi_cpu_gpu_temperature_03549026 - GPU Temperature: 63.4
[image]

The Functions I've set up that trigger off sensor data are also not running.

Possibly all related to a read permissions issue with the DB?

K1rdro, 2023-02-13:
I've also followed the recommendations from the forum:
https://forum.radicaldiy.com/t/zero-values-in-live-measurements/1254

kizniche, 2023-02-13:
What version influxdb did you select during the install?

kizniche, 2023-02-13:
>HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.6.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Mon, 13 Feb 2023 19:37:22 GMT', 'Content-Length': '62'})

It looks like you're trying to use influxdb 2.x, which can only run on 64-bit operating systems. Are you running a 64-bit OS?

K1rdro, 2023-02-13:
Yes, I'm using 2.x, and the 64-bit Raspberry Pi OS.

K1rdro, 2023-02-13:
[image]

kizniche, 2023-02-14:
Did you install this version, or upgrade to this version? If a fresh install, can you provide the setup.log from the install directory?

K1rdro, 2023-02-14:
I selected this version during the install. It's a fresh install, but it didn't generate the log file in the install directory. I'll do an uninstall and run it again to see if I can get it to generate.

K1rdro, 2023-02-14:
Yes, set branch to master and upgrade was what I'd done in my initial troubleshooting per the linked forum post.

Initially I'd installed it with 2.x, and ran into the issue of it not displaying the values. I flashed a new OS, and then tried an install with 1.x instead. It failed to connect during the localhost check at the end, so I ran the installer again with 2.x. The log shows both installs:
[setup.log](https://github.com/kizniche/Mycodo/files/10735283/setup.log)

If you would like me to do a clean OS flash and 2.x install, it's not really a hassle, so please let me know.

closed 2025-10-08 by kizniche