kizniche/Mycodo #1280: influxdb_client.rest.ApiException (401) unauthorized access

dylandn, 2023-02-26:
Just updated from 8.15.4 to 8.15.6, I also ran an apt upgrade (Bullseye 64bit on rpi 4), and now influxdb is broken. No mqtt data inputs seem to be stored, no data is shown in the Live Data page. I know the mqtt input is being generated and reaching the mosquitto server with an phone mqtt client.

Mycodo log is filled with:

```
STOP right now, and please first look to see i2023-02-25 23:18:19,552 - ERROR - mycodo.controllers.controller_pid_eb998d22 - Exception while reading measurement from the influxdb database
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controllers/controller_pid.py", line 439, in get_last_measurement_pid
    last_measurement = read_influxdb_single(
  File "/var/mycodo-root/mycodo/utils/influx.py", line 362, in read_influxdb_single
    data = query_string(
  File "/var/mycodo-root/mycodo/utils/influx.py", line 291, in query_string
    ret_value = query_flux(
  File "/var/mycodo-root/mycodo/utils/influx.py", line 276, in query_flux
    tables = client.query_api().query(query)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/client/query_api.py", line 203, in query
    response = self._query_api.post_query(org=org, query=self._create_query(query, self.default_dialect, params),
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/service/query_service.py", line 279, in post_query
    (data) = self.post_query_with_http_info(**kwargs)  # noqa: E501
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/service/query_service.py", line 305, in post_query_with_http_info
    return self.api_client.call_api(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 343, in call_api
    return self.__call_api(resource_path, method,
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 121, in __call_api
    self._signin(resource_path=resource_path)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 657, in _signin
    http_info = SigninService(self).post_signin_with_http_info()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/service/signin_service.py", line 74, in post_signin_with_http_info
    return self.api_client.call_api(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 343, in call_api
    return self.__call_api(resource_path, method,
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 173, in __call_api
    response_data = self.request(
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/api_client.py", line 388, in request
    return self.rest_client.POST(url,
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/rest.py", line 311, in POST
    return self.request("POST", url,
  File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/_sync/rest.py", line 261, in request
    raise ApiException(http_resp=r)
influxdb_client.rest.ApiException: (401)
Reason: Unauthorized
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.6.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Sun, 26 Feb 2023 07:18:19 GMT', 'Content-Length': '62'})
HTTP response body: {
	"code": "unauthorized",
	"message": "unauthorized access"

```

influxdb is v2.6.1
It's possible the apt upgrade on the side messed things up, hope I can fix this without a reinstall.

dylandn, 2023-02-26:
Diagnosing, I'm not sure this is normal?
```
dietpi@grow:~$ influxd recovery user list
2023-02-26T18:36:21.849854Z	info	Resources opened	{"log_id": "0gFl1RrG000", "system": "bolt-kvstore", "path": "/home/dietpi/.influxdbv2/influxd.bolt"}
ID	Name
dietpi@grow:~$ 
```

dylandn, 2023-03-22:
Something's still broken. 
I've decided to wipe my mycodo setup and reinstall fresh. 
After a fresh reinstall, I immediately set up two MQTT inputs, with debug enabled, and the log is filled with:
```
2023-03-21 23:19:41,791 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Received message: topic: canopy/temperature1, payload: 15.7
2023-03-21 23:19:41,791 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Found channel 0 with topic 'canopy/temperature1'
2023-03-21 23:19:41,792 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Payload represents a float: 15.7
2023-03-21 23:19:41,795 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Adding measurement to influxdb: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 15.7, 'timestamp_utc': datetime.datetime(2023, 3, 22, 6, 19, 41, 791757)}}
2023-03-21 23:19:44,254 - DEBUG - mycodo.inputs.mqtt_paho_e7d45a83 - Received message: topic: canopy/humidity1, payload: 61.9
2023-03-21 23:19:44,255 - DEBUG - mycodo.inputs.mqtt_paho_e7d45a83 - Found channel 0 with topic 'canopy/humidity1'
2023-03-21 23:19:44,256 - DEBUG - mycodo.inputs.mqtt_paho_e7d45a83 - Payload represents a float: 61.9
2023-03-21 23:19:44,263 - DEBUG - mycodo.inputs.mqtt_paho_e7d45a83 - Adding measurement to influxdb: {0: {'measurement': 'humidity', 'unit': 'decimal', 'value': 61.9, 'timestamp_utc': datetime.datetime(2023, 3, 22, 6, 19, 44, 255702)}}
2023-03-21 23:19:44,264 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Received message: topic: canopy/temperature1, payload: 15.8
2023-03-21 23:19:44,270 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Found channel 0 with topic 'canopy/temperature1'
2023-03-21 23:19:44,276 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Payload represents a float: 15.8
2023-03-21 23:19:44,287 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Adding measurement to influxdb: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 15.8, 'timestamp_utc': datetime.datetime(2023, 3, 22, 6, 19, 44, 270410)}}
2023-03-21 23:19:46,768 - DEBUG - mycodo.inputs.mqtt_paho_e7d45a83 - Received message: topic: canopy/humidity1, payload: 61.6
2023-03-21 23:19:46,769 - DEBUG - mycodo.inputs.mqtt_paho_e7d45a83 - Found channel 0 with topic 'canopy/humidity1'
2023-03-21 23:19:46,769 - DEBUG - mycodo.inputs.mqtt_paho_e7d45a83 - Payload represents a float: 61.6
2023-03-21 23:19:46,775 - DEBUG - mycodo.inputs.mqtt_paho_e7d45a83 - Adding measurement to influxdb: {0: {'measurement': 'humidity', 'unit': 'decimal', 'value': 61.6, 'timestamp_utc': datetime.datetime(2023, 3, 22, 6, 19, 46, 769001)}}
2023-03-21 23:19:46,777 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Received message: topic: canopy/temperature1, payload: 15.8
2023-03-21 23:19:46,784 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Found channel 0 with topic 'canopy/temperature1'
2023-03-21 23:19:46,790 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Payload represents a float: 15.8
2023-03-21 23:19:46,800 - DEBUG - mycodo.inputs.mqtt_paho_38738d63 - Adding measurement to influxdb: {0: {'measurement': 'temperature', 'unit': 'C', 'value': 15.8, 'timestamp_utc': datetime.datetime(2023, 3, 22, 6, 19, 46, 784217)}}
and so on
```
Yet in Live Measurements, both inputs show "No Data Last 30 sec", even several minutes later. Asynchronous graph shows no data stored, and a dashboard with  Measurement widgets for both inputs show max age exceeded. 

The web log is filled with:
```
Mar 21 23:24:54 grow python[5093]: 2023-03-21 23:24:54,387 URL for 'last_data' raised and error: (401)
Mar 21 23:24:54 grow python[5093]: Reason: Unauthorized
Mar 21 23:24:54 grow python[5093]: HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.6.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Wed, 22 Mar 2023 06:24:54 GMT', 'Content-Length': '62'})
Mar 21 23:24:54 grow python[5093]: HTTP response body: {
Mar 21 23:24:54 grow python[5093]:         "code": "unauthorized",
Mar 21 23:24:54 grow python[5093]:         "message": "unauthorized access"
Mar 21 23:24:54 grow python[5093]: }
Mar 21 23:24:54 grow python[5093]: Traceback (most recent call last):
Mar 21 23:24:54 grow python[5093]:   File "/home/mycodo/Mycodo/mycodo/mycodo_flask/routes_general.py", line 325, in last_data
Mar 21 23:24:54 grow python[5093]:     data = query_string(
Mar 21 23:24:54 grow python[5093]:   File "/home/mycodo/Mycodo/mycodo/utils/influx.py", line 296, in query_string
Mar 21 23:24:54 grow python[5093]:     ret_value = query_flux(
Mar 21 23:24:54 grow python[5093]:   File "/home/mycodo/Mycodo/mycodo/utils/influx.py", line 281, in query_flux
Mar 21 23:24:54 grow python[5093]:     tables = client.query_api().query(query)
Mar 21 23:24:54 grow python[5093]:   File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/client/query_api.py", line 203, in query
Mar 21 23:24:54 grow python[5093]:     response = self._query_api.post_query(org=org, query=self._create_query(query, self.default_dialect, params),
Mar 21 23:24:54 grow python[5093]:   File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/service/query_service.py", line 279, in post_query
Mar 21 23:24:54 grow python[5093]:     (data) = self.post_query_with_http_info(**kwargs)  # noqa: E501
Mar 21 23:24:54 grow python[5093]:   File "/var/mycodo-root/env/lib/python3.9/site-packages/influxdb_client/service/query_service.py", line 305, in post_query_with_http_info
Mar 21 23:24:54 grow python[5093]:     return self.api_client.call_api(
[... cut ...]
Mar 21 23:24:54 grow python[5093]: HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.6.1', 'X-Platform-Error-Code': 'unauthorized', 'Date': 'Wed, 22 Mar 2023 06:24:54 GMT', 'Content-Length': '62'})
Mar 21 23:24:54 grow python[5093]: HTTP response body: {
Mar 21 23:24:54 grow python[5093]:         "code": "unauthorized",
Mar 21 23:24:54 grow python[5093]:         "message": "unauthorized access"
Mar 21 23:24:54 grow python[5093]: }```

kizniche, 2023-03-22:
Install with influxdb 1.x

dylandn, 2023-03-22:
That did the trick, thank you!
Influxdb 2 not quite ready for primetime?

dylandn, 2023-03-22:
No, 64bit Debian 11 on an rpi4

kizniche, 2023-03-22:
You can try installing Mycodo from the master branch to see if 2.x will work

kizniche, 2023-04-06:
I believe this is fixed with the latest release. If you find this issue persists, feel free to reopen this issue.

closed 2023-04-06 by kizniche