kizniche/Mycodo #1281: Peristaltic Pump: PCF8574, all pump activity is shown as CH0 activity in graph when regulate ph and EC function uses pumps

Wolkar10, 2023-03-03:
### Describe the problem/bug

Thank you very much for this excellent software. I have been testing and playing it for a while before putting my system in prduction use and I noticed an issue with graph. When Regulate pH and Electrical Conductivity function uses different peristaltic pumps (Peristaltic Pump: PCF8574 8-Channel I/O Expander) all active times or pumped milliliters are shown in graph as channel 0 activity. This happens only with regulate ph and EC function, if I manually activate different pumps or create a function that uses different pumps periodically graph shows times and amounts correctly to right channels. If you need more info I'm more than happy to help. I also have a little wish for future development - it would be nice if you could name different pumps in Peristaltic Pump: PCF8574 8-Channel I/O Expander module, it would make graphs and output page more informative, but this is just one nice to have feature.

### Versions:

 - Mycodo Version: 8.15.7
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: Linux raspberrypi 5.15.84-v7l+ #1613

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1.  Add different channels of Peristaltic Pump: PCF8574 8-Channel I/O Expander as outputs for regulate ph and EC function
2.  Create graps where ph up and down + nutrient pumps activities are shown.
3.  Cause condition where function adjusts ph up and down or regulates EC with different pumps
4.  All activity is shown as channel 0 activity (first pump)

### Expected behavior

Pump activity is shown correctly when pumps are manually turned on or when my own test function turns them on. Problem probably happens only with regulate ph and EC function.

kizniche, 2023-03-04:
I just committed a potential fix for this. If you'd like to [upgrade](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and test, let me know if the issue persists.

Wolkar10, 2023-03-05:
Hi, thank you very much :) I tested all possible cases I could figure out. Regulate pH and EC function and own test function work now fine, both times and dispenced volumes are visible correctly in graph. I noticed that if you use pumps manually from output page button or send duration command to controller from output configuration page those actions are visible in graph only as duration and nothing is shown as dispensed volume. I can't remember did this work differently before I took the fix in use. My plan was to use only dispensed volumes in graph so I just have to remember that manual actions are not visible in that view, those are exception to normal operation anyway.

kizniche, 2023-03-05:
Thanks for the feedback. I'll look into that other issue.

kizniche, 2023-03-05:
It seems what you described is the expected behavior. It is not straightforward to allow saving volume when a duration is sent to the output. I'll think about if this can be done and if it's worth putting in the time to implement.

Wolkar10, 2023-03-07:
Thank you. This is not an issue to me now when I know how the system/graph works with manual actions.

closed 2023-08-21 by kizniche