kizniche/Mycodo #1283: Successful dependency install not reflected in UI

cDurham, 2023-03-04:
If you are going to post a new issue, next read How to Write a Good Bug Report at https://forum.radicaldiy.com/t/how-to-write-a-good-bug-report/71

Please complete as many of the sections below, if applicable, to provide the most information that may help with investigating your issue. Replace the text in brackets with your own text describing the issue. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help those providing assistance by reducing the time required to help you.

### Describe the problem/bug

Hi friends! Thanks for the amazing software.

I am on a fresh install of Mycodo. When installing a depdency for various outputs/inputs, the log shows success, but the front end still says dependency missing and attempts to reinstall a dependency

This occurs on different depdencies, pylibftdi==0.20.0, pigpio==1.78. I didn't try a bunch of other things, but the problem is the same, successful install, no recognition of those depdencies.

I checked the .dependency but the file is empty. (not sure what SHOULD be in there, but I expected something)

### Versions:

 - Mycodo Version: 8.15.7
 - Raspberry Pi Version: 4B
 - Raspbian OS Version: 10

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

on a freshly flashed pi
install mycodo 8.15.x and then upgrad to latest (I inititally installed a lower version, doubt this matters)
try to add 'PWM: Raspberry Pi GPIO'
install the dependencies, 

### Expected behavior

I expected the successful dependency install to reflect in the ui

### Screenshots

<details>
<summary>Here are some example logs</summary>
[2023-03-04 12:50:19] Dependency installation beginning. Installing: pylibftdi

[2023-03-04 12:50:19] 
[2023-03-04 12:50:19] #### Installing/updating pylibftdi==0.20.0 (pip-pypi)

[2023-03-04 12:50:20] End install of pylibftdi

[2023-03-04 12:50:20] 
[2023-03-04 12:50:20] #### Setting permissions

[2023-03-04 12:50:21] #### Dependencies installed. Restarting frontend and backend...

[2023-03-04 12:50:24] #### Dependency install complete.

[2023-03-04 14:22:20] Dependency installation beginning. Installing: pigpio

[2023-03-04 14:22:20] 
[2023-03-04 14:22:20] #### Installing/updating pigpio==1.78 (pip-pypi)

[2023-03-04 14:22:20] End install of pigpio

[2023-03-04 14:22:20] 
[2023-03-04 14:22:20] #### Setting permissions

[2023-03-04 14:22:22] #### Dependencies installed. Restarting frontend and backend...

[2023-03-04 14:22:25] #### Dependency install complete.
</details>

kizniche, 2023-03-04:
Your logs indicate nothing is installed. A successful pypi dependency install looks like this:

```
[2022-09-09 19:03:24] Dependency installation beginning. Installing: MCP342x

[2022-09-09 19:03:24] 
[2022-09-09 19:03:24] #### Installing/updating MCP342x==0.3.4 (pip-pypi)
[2022-09-09 19:03:25] Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
[2022-09-09 19:03:26] Collecting MCP342x==0.3.4
[2022-09-09 19:03:26]   Downloading https://www.piwheels.org/simple/mcp342x/MCP342x-0.3.4-py2.py3-none-any.whl (6.5 kB)
[2022-09-09 19:03:26] Requirement already satisfied: smbus in ./env/lib/python3.9/site-packages (from MCP342x==0.3.4) (1.1.post2)
[2022-09-09 19:03:29] Installing collected packages: MCP342x
[2022-09-09 19:03:29] Successfully installed MCP342x-0.3.4

[2022-09-09 19:03:29] End install of MCP342x

[2022-09-09 19:03:29] 
[2022-09-09 19:03:29] #### Setting permissions

[2022-09-09 19:03:32] #### Dependencies installed. Restarting frontend and backend...

[2022-09-09 19:03:38] #### Dependency install complete.
```

cDurham, 2023-03-04:
Hmm interesting. I tried manually installing pigpio==1.78

I navigated to ~/Mycodo/env/bin and ran `pip install -v "pigpio==1.78"`
I got the following output: 
```
Using pip 23.0.1 from /usr/local/lib/python3.7/dist-packages/pip (python 3.7)
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pigpio==1.78 in /usr/local/lib/python3.7/dist-packages (1.78)
```

Not sure what to try next. Maybe reinstall mycodo?

kizniche, 2023-03-04:
I've never seen your issue and don't have any recommendations except to reflash Raspberry Pi OS and start over.

cDurham, 2023-03-04:
Thanks. fingers crossed! feel free to close :)

cDurham, 2023-03-04:
Ok nuking the Mycodo directory and reinstalling worked. no clue why, but we're off to the races!

Thanks again

closed 2023-03-04 by kizniche