kizniche/Mycodo #1287: Error Upgrading

stardawg, 2023-03-20:
Version: 8.14.2
Database: 07c7c8ebc195
Model: Raspberry Pi 4 Model B Rev 1.5

Release:
Distributor ID: Debian
Description: Debian GNU/Linux 11 (bullseye)
Release: 11
Codename: bullseye

Firmware:
b''

Error (Full Traceback):

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 277, in decorated_view
    return current_app.ensure_sync(func)(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_restx/api.py", line 672, in error_router
    return original_handler(e)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/var/mycodo-root/env/lib/python3.9/site-packages/flask_login/utils.py", line 279, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 534, in admin_upgrade
    current_latest_major_version = current_latest_release.split('.')[0]
AttributeError: 'NoneType' object has no attribute 'split'

kizniche, 2023-03-20:
Duplicate of #1266

Everyone needs to start using the search feature before opening an issue. There are literally a dozen of this duplicate issue I've had to close.

kizniche, 2023-03-20:
The very first line of the bug report template instructs to perform a search:

*STOP right now, and please first look to see if the issue you're about to submit is already an open or recently closed issue at https://github.com/kizniche/Mycodo/issues*